bblanchon/ArduinoJson #1273: Key and value is present but returns NULL

vaz83, 2020-05-19:
Hi,
I am facing a strange problem. I saved a JSON document into the ESP8266 file system.

Then i load it using this code:
`  DynamicJsonDocument jDocLoad(1500);
  DynamicJsonDocument jDoc(1500);

  if (LittleFS.begin()) 
  {
    if (LittleFS.exists("/topRecordings.json")) 
    {
      File cFile = LittleFS.open("/topRecordings.json", "r");
      if (cFile)
      {
         size_t size = cFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);
        cFile.readBytes(buf.get(), size);
        DeserializationError err = deserializeJson(jDocLoad,buf.get());
        Serial.println(err.c_str());
        serializeJsonPretty(jDocLoad,Serial);
      }
      cFile.close();
  }`

The output of the serializeJsonPretty is this:

{
  "maxTempC": 26.54,
  "maxTempF": 79.772,
  "maxTempTimeStampUTC": "2020-05-19T15:51:44",
  "maxTempUNIXTime": 1589903504
}

After, i try to get the maxTempC value, but gives no value and when i put the line jDocLoad.containsKey("maxTempC") it returns FALSE, while the key and value is there which can be seen on the serializeJSonPretty. 

What can be the problem?

ewaldc, 2020-05-19:
You can deserialize directly from a stream:
`DeserializationError err = deserializeJson(jDocLoad, cFile);`
Less lines of code and less memory used...

What is your problem exactly ? If _serializeJsonPretty_ returns the correct Json, than it means you have no error from _deserializeJson_ and thus all is read fine from LittleFS. In that case, the problem is not in the code above.  Or do I understand this wrong?

vaz83, 2020-05-19:
It returns the correct code, but right after i do this:

jDocLoad.containsKey("maxTempC") which returns false.. it's only the first field, i don't understand why..

vaz83, 2020-05-19:
Deserialize directly from the stream solved the problem! :) thanks

closed 2020-05-19 by vaz83