bblanchon/ArduinoJson #1274: unable to deserialize a json array without ',' at the end

ilyasamraoui, 2020-05-20:
hello,
I tried to deserialize this json array "[1,2,3]" with the code below:
`const size_t CAPACITY = JSON_ARRAY_SIZE(3);
StaticJsonDocument<CAPACITY> doc;
deserializeJson(doc, "[1,2,3]");
`
and it didn't succeed but it worked when i changed it to :

`deserializeJson(doc, "[1,2,3,]");
`
I'm using the esp32 with arduinoJson 6.15.2

bblanchon, 2020-05-21:
Hi @ilyasamraoui,

I'm pretty sure you made a mistake.
What is your criterion to say that "it worked"?

Best regards,
Benoit

ilyasamraoui, 2020-05-21:
Hello @bblanchon 

I'm sorry i didn't give more details here is the real code:
`const size_t CAPACITY = JSON_ARRAY_SIZE(3);
    StaticJsonDocument<CAPACITY> doc;
    DeserializationError success = deserializeJson(doc, "[1,2,3]");
    if(!success) {
        Serial.println("invalid");
    }`

when i execute this code it prints invalid but it actually does the parsing i can read the values with this code :
`JsonArray array = doc.as<JsonArray>();
    for(JsonVariant v : array) {
        Serial.println(v.as<int>());
    }`

when i change the code to :
`DeserializationError success = deserializeJson(doc, "[1,2,3,]");`

it doesn't print invalid so i guess the problem is in the DeserializationError

Best regards,
Ilyas

bblanchon, 2020-05-22:
Hi @ilyasamraoui,

The mistake becomes obvious when you rename the variable `success` to `error`.

Best regards,
Benoit

ilyasamraoui, 2020-05-24:
hello @bblanchon ,

im really sorry for wasting your time that's correct i had a different use case that worked with the variable called success that's why it confused me here but i fixed it now thank you so mush and great work on the library full respect

Best regards,
Ilyas

bblanchon, 2020-05-25:
No problem :-)
Thank you for using ArduinoJson.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink:

closed 2020-05-25 by bblanchon