bblanchon/ArduinoJson #1275: esp32 crashes if invlaid field is been referenced

az-z, 2020-05-22:
Hello,
`truct WIFI_Config {
	char ssid[10] ;
	char password[10] ;
	char hostname[10] ; 
};

WIFI_Config config;`
...
bool loadConfiguration(const char* fname /*= filename */ , WIFI_Config &cfg /*= config */) {
	SPIFFS.begin();
	File file = SPIFFS.open(fname,"r");
	StaticJsonDocument<200>  doc;
	DeserializationError error = deserializeJson(doc, file);

	strlcpy(cfg.ssid,doc["ssid"],sizeof(cfg.ssid));
->	strlcpy(cfg.password, doc["password"],sizeof(cfg.password) );
->     strlcpy(cfg.password, doc["pass"],sizeof(cfg.password) );
	Serial.println("LoadCofng: copied password");
...
`
an attempt to extract/reference "pass" ( instead of "password") crashes exp32.

ewaldc, 2020-05-22:
Hmm, does _strlcpy_ not require that the source string must be a valid null terminated string (an not a nullptr)?  If _doc["pass"]_ would return something like "NULL" it shouldn't crash, but then one would need to check the return value of strlcpy to find that it would not have copied 11 characters (= _sizeof(cfg.password)_)

bblanchon, 2020-05-22:
Hi @az-z,

As @ewaldc pointed out, it's not ArduinoJson that crashes; it's `strcpy()`.
See an explanation here:

* [Why does my device crash or reboot?](https://arduinojson.org/v6/faq/why-does-my-device-crash-or-reboot/)

Best regards,
Benoit

az-z, 2020-05-22:
Gentlemen,  thank you. I was wrong in my assumption. Thank you for pointing
it out.
Please close, as not a bug.

closed 2020-05-22 by bblanchon