bblanchon/ArduinoJson #1278: ERROR: Filter is not a member of ArduinoJSON

MHz000, 2020-05-28:
compiling an unchanged JsonFilterExample from ArduinoJson.h yields in:
"Filter is not a member of ArduinoJson " 
regardless of choosen board ESP8266 or ESP32 (Arduino latest IDE)

Thank you for any advise

bblanchon, 2020-05-28:
Hi @MHz000,

An older version of ArduinoJson is probably installed on your computer.
Look for a string like this in the compilation output:

    Using library ArduinoJson at version 6.15.2

`DeserializationOption::Filter` was introduced in [ArduinoJson 6.15.0](https://arduinojson.org/news/2020/03/22/version-6-15-0/).

Best regards,
Benoit

MHz000, 2020-05-29:
Benoit thank you for the hint. It was exactly the case😂

All the best
MHz000

closed 2020-05-29 by bblanchon