bblanchon/ArduinoJson #1279: Need to convert a Json doc to url encode

haeleng, 2020-05-29:
Hi,

First I'm not a software engineer but I like to write some code for my hobby.
So perhaps my question is simple.
I use a Arduino Mega2560 with Visual Studio for a project which sends data to a PTZ camera via HTTP to control it remotely.
I have made a JSON doc with the library you created. Thanks for that.
However it seems that the PTZ camera only works when I send the JSON doc in a url encoded way.
I have a url encoded subroutine which accepts a string as input.
Question: How can I convert the JSON doc so that it is url encoded.
Thanks

bblanchon, 2020-05-30:
Hi @haeleng,

I'm sorry, but I don't understand the question.
To me, one uses either URL encoding or JSON, not both.
Can you provide an example?

ArduinoJson doesn't support URL encoding, and there is no plan to add this in the future.
You can probably compose your URL-encoded string manually by using sprintf().

Best regards,
Benoit

ewaldc, 2020-05-30:
From the top of my head, typed on my phone in the train (thus not compiled or checked for syntax errors)

```
char json[] = "{\"first\":\"number1\",\"second\":\"number2\"}";
DynamicJsonDocument doc(1024);
deserializeJson(doc, json);
JsonObject root = doc.as<JsonObject>();

bool first = true;
String url;
for (JsonPair kv : root) {
   url += (first) ? (first = false; F("?")) : F("&");
   url += kv.key();
   url += F("=");
   url += urlEncode(kv.value().as<const char*>());
 }
```

closed 2020-06-16 by bblanchon