bblanchon/ArduinoJson #1280: Sort a JsonArray by JsonVariant properties

alfo, 2020-05-29:
I'm starting with a JsonArray generated from the following JSON using `JsonArray array = json.as<JsonArray>();`

```json
[	
  	{"timeToStation": 330, "towards": "Kennington via CX"},
	{"timeToStation": 180, "towards": "Kennington via CX"},
	{"timeToStation": 270, "towards": "Morden via Bank"}
]
```

I'm trying to find a way to sort these items by the value of "timeToStation". Is there a way to do this built in to ArduinoJson?

I've tried the following code with no success:

```c
for (size_t i = 1; i < array.size(); i++) {
      
  for (size_t j = i; j > 0 && (array[j-1]["timeToStation"].as<int>() > array[j]["timeToStation"].as<int>();); j--) {
    JsonVariant tmp = array[j-1];
    array[j-1] = array[j];
    array[j] = tmp;
  }
      
}
```

This sometimes generates empty array elements and sometimes crashes the program altogether. Would anyone mind helping me get this working?

Platform: ESP32

bblanchon, 2020-05-29:
Hi @alfo,

Unfortunately, there is no efficient way to do that directly with [`JsonArray`](https://arduinojson.org/v6/api/jsonarray/):

1. You cannot swap values like that because [`JsonVariant`](https://arduinojson.org/v6/api/jsonvariant/)s are references (the assignment changes the pointer, not the value)
2. Trying to swap values with [`JsonVariant::set()`](https://arduinojson.org/v6/api/jsonvariant/set/) fails as well because you'll erase the original value.
3. You cannot create the `tmp` value because a [`JsonVariant`](https://arduinojson.org/v6/api/jsonvariant/) must be attached to a [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/), but you can use a [`StaticJsonDocument`](https://arduinojson.org/v6/api/staticjsondocument/) as a temporary storage.
4. When you assign a new value via [`JsonVariant::set()`](https://arduinojson.org/v6/api/jsonvariant/set/), it makes a copy, which consumes more room in the original  [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/), and this repeats for each iteration of the bubble sort algorithm.

Here is a demo that overcomes all these challenges:
https://wandbox.org/permlink/Kk62zq7QKsYch3ox

As you can see, it sorts correctly, but [`JsonDocument::memoryUsage()`](https://arduinojson.org/v6/api/jsondocument/memoryusage/) grows from 406 to 820.
It may be acceptable here because we only have 3 items, but remember that the bubble sort algorithm has a complexity of n².

Instead of fighting against the design of ArduinoJson, I strongly recommend that you extract an array, a [vector](https://en.cppreference.com/w/cpp/container/vector), or a [list](https://en.cppreference.com/w/cpp/container/list), and call [`std::sort`](https://en.cppreference.com/w/cpp/algorithm/sort).

You can see a demo here:
https://wandbox.org/permlink/M21V0E1K6bXaefyr

Note that we can avoid the [`std::string`](https://en.cppreference.com/w/cpp/string/basic_string) if we use the "zero-copy" mode:
https://wandbox.org/permlink/oa9wjl6Lzk4NgBE7

I understand that you can be disappointed that ArduinoJson doesn't offer a better way to do that, but remember that ArduinoJson is a *serialization* library, and *sorting* is not part of it.
Moreover, I had to make many compromises to make the code efficient in size, speed, and memory consumption.

Best regards,
Benoit

alfo, 2020-05-29:
Thank you for your quick and detailed reply! Those examples look perfect, and I understand why that functionality isn’t offered inside ArduinoJson now.

bblanchon, 2020-05-30:
You're welcome @alfo.
Good luck with your project!

closed 2020-05-30 by bblanchon