bblanchon/ArduinoJson #1281: ESP8266 got a Soft WDT reset

yuan910715, 2020-05-31:
Hello, bblanchon
Thank you for developing such a useful library
I ran into a problem and spent many days without solving it, so I had to ask you for help

I'm using ESP8266(nodemcu) and this is a minimum code:  

```
String test::doUpdateCurr(String key,String location) {  
    std::unique_ptr<BearSSL::WiFiClientSecure>client(new BearSSL::WiFiClientSecure);     
    client->setInsecure();    
    HTTPClient https;  
    String url="XXXXXXXXXXXXXXXX;  
    Serial.println(url);  
    if (https.begin(*client, url)) {  // HTTPS    
      // start connection and send HTTP header  
      int httpCode = https.GET();    
      // httpCode will be negative on error  
      if (httpCode > 0) {     
        if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {  
          String payload = https.getString();  
          Serial.println(payload);  
          StaticJsonDocument<1024> doc;
          deserializeJson(doc,payload);
         String a =doc["XXXXX"];  
           return a;      
        }  
      } else {    
        Serial.printf("[HTTPS] GET... failed, error: %s\n", https.errorToString(httpCode).c_str());               
        return "-0";               
      }    
      https.end();  
    } else {             
      Serial.printf("[HTTPS] Unable to connect\n");   
      return "-0";   
    }  
} 
```

I will call this function every minute , but sometimes ESP8266 will restart
```
Soft WDT reset
22:44:02.072 -> 
22:44:02.072 -> >>>stack>>>
22:44:02.072 -> 
22:44:02.072 -> ctx: sys
22:44:02.072 -> sp: 3ffff6c0 end: 3fffffb0 offset: 01a0
22:44:02.072 -> 3ffff860:  00000000 00000000 3ffeffd4 40203845  
22:44:02.072 -> 3ffff870:  03d71401 00000000 3ffeffd4 40208d82  
22:44:02.072 -> 3ffff880:  00000000 00000000 3fff19dc 00000008  
22:44:02.072 -> 3ffff890:  00003a98 00051fdd 3ffefe44 40204adf  
22:44:02.072 -> 3ffff8a0:  000000e3 00000005 81dd1f23 00000000  
22:44:02.072 -> 3ffff8b0:  00000001 000346f3 3fff19dc 40222aca  
22:44:02.106 -> 3ffff8c0:  3fff1edc 3ffef56c 3fff19dc 00000001  
22:44:02.106 -> 3ffff8d0:  00000001 00000001 3ffefe44 40204dd0  
22:44:02.106 -> 3ffff8e0:  3ffef56c 00000000 3ffefe44 40204fb3  
22:44:02.106 -> 3ffff8f0:  00000012 3ffee80c 00000001 40208d18  
22:44:02.106 -> 3ffff900:  00000000 00000000 00000000 402094f2  
22:44:02.106 -> 3ffff910:  00000000 3ffefe44 3ffeffd4 40203ba0  
22:44:02.106 -> 3ffff920:  3ffee704 00000d50 3ffff998 00000000  
22:44:02.106 -> 3ffff930:  000001bb 3ffefe44 3ffef56c 00000000  
22:44:02.140 -> 3ffff940:  000001bb 3ffefe44 3ffef56c 402050d1  
22:44:02.140 -> 3ffff950:  4020b7d0 f2926927 4020b7d0 f2926927  
22:44:02.140 -> 3ffff960:  3ffefe44 3ffffe38 3ffffe2c 40206339  
22:44:02.140 -> 3ffff970:  ffffffff 3ffffe58 3ffffe2c 40206ac1  
22:44:02.140 -> 3ffff980:  70747400 00000073 80fff9e0 00000000  
22:44:02.140 -> 3ffff990:  00000000 00000000 3ffffe2c 40206f87  
22:44:02.140 -> 3ffff9a0:  80fe87d2 3ffef700 0051006f 8020895d  
22:44:02.173 -> 3ffff9b0:  0000003a 0000004f 3ffff9e0 40207e55  
22:44:02.173 -> 3ffff9c0:  3ffe8859 00000001 00000000 40207e74  
22:44:02.173 -> 3ffff9d0:  3fffff20 00000000 3ffffe2c 40206e34  
[... 82 lines cut ...]
22:44:02.553 -> 3fffff00:  3ffef3f4 0000000b 3fffff70 40207feb  
22:44:02.553 -> 3fffff10:  0000000b 00000000 4bc6a7f0 00000000  
22:44:02.553 -> 3fffff20:  3ffef5dc 006f006f 00ffff70 40208034  
22:44:02.553 -> 3fffff30:  3ffef3f4 3ffee6d4 3ffefe44 40208218  
22:44:02.553 -> 3fffff40:  3fffff70 00000000 3fffff70 40208248  
22:44:02.553 -> 3fffff50:  3ffee6d4 000001f4 3ffee7b0 3ffee7b0  
22:44:02.587 -> 3fffff60:  3fffdad0 00000000 3ffe8524 4020247b  
22:44:02.587 -> 3fffff70:  3ffef94c 000b000f 00000009 3ffef914  
22:44:02.587 -> 3fffff80:  0020002f 00000000 00000001 40100154  
22:44:02.587 -> 3fffff90:  3fffdad0 00000000 3ffee770 40100175  
22:44:02.587 -> 3fffffa0:  feefeffe feefeffe 3ffee770 40208e30  
22:44:02.587 -> <<<stack<<<
22:44:02.622 -> 
22:44:02.622 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
22:44:02.622 -> 
22:44:02.622 -> load 0x4010f000, len 3456, room 16 
22:44:02.622 -> tail 0
22:44:02.622 -> chksum 0x84
22:44:02.622 -> csum 0x84
22:44:02.622 -> va5432625
22:44:02.622 -> ~ld
```

Esp8266 get a soft WDT reset  maybe 10 minutes, maybe 1 hour
I really don't know why, could you help me? thanks.

bblanchon, 2020-05-31:
Hi @yuan910715,

ESP8266 is very stingy with stack memory, so this could be a stack overflow.
Can you try with a [DynamicJsonDocument](https://arduinojson.org/v6/api/dynamicjsondocument/) ?

See also:

* [Why does my device crash or reboot?](https://arduinojson.org/v6/faq/why-does-my-device-crash-or-reboot/)

Best regards,
Benoit

yuan910715, 2020-06-01:
@bblanchon 
I tried but failed again .
But I change the url to another **http** url ,I run it for 5 hours,not WDR reset , I think the issue maybe caused by BearSSL library when I use HTTPS, I will tell you when I solve it ,thanks.

ewaldc, 2020-06-01:
_https.getString();_ is known to cause WDT reset as it is a blocking call.  Some things to try...
Skip the use of an intermediate string as it results in lots of memory copying making things worse.
Let AJ read directly from the _https_ client stream and set a timeout.
Set your serial console to 115200 baud (below is known to cause WDT reset)
Also check if your are dealing with chunked encoding.
BearSSL makes it all even more difficult to get it crash free.  I managed to keep it 7 days alive but have to do a software restart every week.
Which LWIP model did you use? Low memory with MSS=536 ?

```
  Serial.begin(115200);
  https.setTimeout(3000);
  // Parse JSON object
  DeserializationError error = deserializeJson(doc, https);
```

yuan910715, 2020-06-03:
@bblanchon 
I found that using bearSSL requires about 25kb of stack memory , and I did some tests and determined that the WDT reset caused by insufficient stack memory , I think the stack memory of ESP8266 is too small, so it is not suitable to use SSL while having complicated programs .
Thanks!
Also thanks to @ewaldc

closed 2020-06-03 by yuan910715