bblanchon/ArduinoJson #1283: Sending a *.ino file in a POST request with ArduinoJson

dcota, 2020-06-01:
Hi!
I was wondering if it is possible to send a *ino file in a POST request using ArduinoJson6. I am dealing with a project where I need to send data and the correspondent code to a server and I want to send everything in a unique request. Or in separate requests in alternative.
Thank you.
Duarte

bblanchon, 2020-06-02:
Hi Duarte,

You can put whatever you want in the request's body, including the text of a file.
However, I don't see how this question relates to ArduinoJson.

Best regards,
Benoit

dcota, 2020-06-02:
Thanks for the reply, I managed to stringify everything I need to send in the same request, because I need to relate the student ID with the data that is being sent to the server for later analytics.
Thank you very much.
Duarte

closed 2020-06-02 by bblanchon