bblanchon/ArduinoJson #1284: How to access badly array by numeric key

alfo, 2020-06-01:
Hi, me again. I'm trying to figure out how to access the elements of this:

```json
{
	"full_count": 8524,
	"version": 4,
	"249b9060": ["780AEC", 51.9198, 0.6374, 82, 10925, 312, "2241", "F-EGVN1", "B77L", "B-2042", 1591013427, "STN", "FRA", "CZ465", 0, 1280, "CSN465", 0, "CSN"]
}
```

Bearing in mind that the string "249b9060" changes on every refresh...

This is what I've tried:

```c
StaticJsonDocument<2048> json;
deserializeJson(json, payload);

JsonArray jsonFlight = json[2];

int altitude = jsonFlight[4].as<int>();
```

but this causes a crash, I guess I can't access it as `json[2]`. I also tried this from issue #226:
```c
StaticJsonDocument<2048> json;
deserializeJson(json, payload);

const char* nodeID = json[2].begin()->key;
JsonArray jsonFlight = json[nodeID];

int altitude = jsonFlight[4].as<int>();
```

But that method doesn't seem to work anymore? It's annoying the JSON is structured so badly but this is coming straight from an API.

ewaldc, 2020-06-02:
How about:
```
JsonArray jsonFlight;
int itemCount = 0;
for (JsonPair kv : json) {
    if (itemCount++ == 2) jsonFlight = kv.value().as<as<JsonArray>();
}
```
Haven't tested it though (written from my phone on the train...)
Assuming it's always the third object in the JSON text of course...

bblanchon, 2020-06-02:
Hi @alfo,

First, you must know that accessing a member by its position is very brittle because the sender is free to send them in any order. JSON doesn't guarantee the order of keys in objects.

If you still want to do that, you can use [JsonObject's iterator](https://arduinojson.org/v6/api/jsonobject/begin_end/):

```c++
auto it = doc.as<JsonObject>().begin();
it += 2;
JsonArray values = it->value();
```

See a complete demo here:
https://wandbox.org/permlink/iyMMMoGabASl8gKV

Lastly, you say you experienced a crash, and I must make sure that it's not ArduinoJson's fault.
Can you double-check that ArduinoJson was the cause of the crash?
If so, please share an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example) as soon as possible.

Best regards,
Benoit

alfo, 2020-06-02:
Hi Benoit, thanks for this. Yeah I'm aware it's not a great solution, but this seems to be the way that Flightradar have decided to structure their data. I've got some other logic that checks that the input is as expected.

No worries about the crash, it turned out to be some unrelated sloppy char array code.

closed 2020-06-02 by alfo