bblanchon/ArduinoJson #1285: JsonDocument* returns NULL

pranjal-joshi, 2020-06-01:
I have written the following method in my class in which I am trying to read JSON file from SPIFFS memory of ESP8266. **(This code compiles successfully, the issue is described after the code!)**
```
SmartFileSystemFlags_t SmartFileSystem::readJsonFile(JsonDocument *doc, const char* fname) {
  if(SPIFFS.begin()) {
    uint16_t i;
    char buf[JSON_BUF_SIZE];
    File f = SPIFFS.open(fname,"r");
    if(!f) {
      if(DEBUG)
         printDebug("ERROR: Failed to open file for reading.");
         return FileOpenError;
    }
    f.readBytes(buf,f.size());
    deserializeJson(*doc, buf);
    if(DEBUG) {
      Serial.print(DBG_FS);
      Serial.print("INFO: Read following data from file ");
      Serial.println(fn);
      Serial.print(DBG_FS);
      Serial.print("DATA: ");
      serializeJson(*doc, Serial);
      Serial.println();
    }
    return FileReadOk;
  }
  else {
    if(DEBUG)
      printDebug("ERROR: Failed to mount SPIFFS.");
    return SPIFFSError;
  }
}

DynamicJsonDocument SmartFileSystem::readConfigFile(void) {
  DynamicJsonDocument doc(JSON_BUF_SIZE);
  readJsonFile(&doc, "/config.json");
  return doc;
}
```
Here, I am able to read the file and even `serializeJson(*doc, Serial);` works fine by displaying the JSON output. But when I call the following from my main code, I always get a null!
```
// main code
DynamicJsonDocument confJson(1024*2);
SmartFileSystem fsys;
confJson = fsys.readConfigFile();
// what I am trying to achieve is as follow:
const char* username = confJson["username"];
```
I think I am messing up with pointers or something while passing args by reference. Please let me know!!

ewaldc, 2020-06-02:
Simplify your code by having AJ read from SPIFFS directly (less memory fragmentation and smaller  code size)
`    deserializeJson(*doc, f);`

Avoid passing a pointer to DynamicJsonDocument, even though it's technically OK.  If you need to read from multiple files, you won't be able to hard-code the size of DynamicJsonDocument.  In that case, you can use something like this:
```
DynamicJsonDocument getJsonDocFromFile(String fileName) {
	File f = SPIFFS.open(fileName, "r");  // add code to test if f exists if needed
	DynamicJsonDocument doc(ESP.getMaxFreeBlockSize() - 4096);
	DeserializationError err = deserializeJson(doc, f, DeserializationOption::NestingLimit(20));
	if (err == DeserializationError::Ok) doc.shrinkToFit();
	else DbgPrintln(F("deserializeJson failed "), err.c_str());
	f.close();
	return doc;
}
```
Another possibility is to use the file size + 25-50%, eventually in a first attempt.

In this way you also don't forget to close the file (and leak memory).   If you use  _getJsonDocFromFile_ from another function, you will be assured that the JsonDocument is properly destroyed after each function call.

pranjal-joshi, 2020-06-02:
@ewaldc 
I got my code by simply **modifying the return type of `SmartFileSystem::readConfigFile(void)` method** as follow:
```
StaticJsonDocument<JSON_BUF_SIZE> SmartFileSystem::readConfigFile(void) {
  DynamicJsonDocument doc(JSON_BUF_SIZE);
  readJsonFile(&doc, CONF_FILE);
  doc.shrinkToFit();
  return doc;
}
```
Why does the return type `DynamicJsonDocument` returns null while `StaticJsonDocument<size>` works? Am I wasting memory by doing so? If yes, how to optimize it?

ewaldc, 2020-06-02:
Don't have access to an ESP system now, but IMHO more code than needed often increases the chance of an error.  
For example, `char buf[JSON_BUF_SIZE];` should be null terminated in the way that _deserializeJson_ is called in the code above, but from the posted code it might not be.  Is `JSON_BUF_SIZE > f.size() ` ?   Impossible to say unless you post the file and full code, but these little things can cause unpredictable behavior.  A better way is not to hard-code these things and let ArduinoJson read from the SPIFFS stream directly.   You will find that it is difficult to do better than Benoit Blanchon, he is one of the best C++ coders around :-)

pranjal-joshi, 2020-06-02:
@ewaldc 
Thanks for great suggestions! I have already experience that if I increase `JSON_BUF_SIZE` too much then esp throws a WDT reset.
I will keep your suggestions in my mind while optimizing the code.
Thanks to the author and all contributors for this awesome library!
Closing this issue. Feel free to reopen any time.

closed 2020-06-02 by pranjal-joshi