bblanchon/ArduinoJson #1286: Serialize a structure on an ESP32

bobcroft, 2020-06-04:
Hi, I have the following structure, can it be serialized in JSON and then deserialized on a second ESP8266 device when sent by MQTT.  If so, how would one do it?  I have read the book and ccannot find an answer there.  I think I may have to send the data as separate arrays.  I am OK with the MQTT part once I get the data into JSON format.

typedef struct     // bank data
{
  bool    opState[NO_0F_HOURS];
  uint16_t onTime [NO_0F_HOURS];
} SETTING_DATA ;

SETTING_DATA bankData   [NO_OF_RELAYS];   // 8 structures - one for each relay

bblanchon, 2020-06-05:
Hi @bobcroft,

This looks like an array of objects, what else do you need?

```json
[
  {"opState":[true, false], "onTime":[12, 35]},
  {"opState":[false, false], "onTime":[15, 64]}
]
```

I'm sorry that you didn't find the answer to your question in my book.
I usually get very positive feedback from readers, but one book cannot suit everybody.
Please write me an email so I can send you a refund.

Best regards,
Benoit

bobcroft, 2020-06-05:
Benoit, firstly I did not mean to imply that the book had failed me, it hasn't and I have learnt a great deal from it.  I think it is a superbly written and informative book.  The very last thing I want is a refund.
Regarding my question, I was asking can the structure be put into Json format as a single entity?  There are 14 opStates and 14 onTimes for each of the 8 relays, so quite a large structure.  Upon reflecting on how the data is used I have decided that it is probably better to use two arrays.  For the opStates I use a bit field structure and  a union to create a uint16_t value called bits, so I can access each bit or the numerical representation.  Thus an array of 8 'bits' values gives me all the 14 * 8 bits values I need.  The project has evolved over time and I sometimes think it is wise to go back and look at the code and change it, if necessary, to reflect improvements, additions etc.
I'll try the standard array method and get back to you if I need more help.
Thank you again for the absolutely brilliant library and accompanying book.

bblanchon, 2020-06-05:
Hi @bobcroft,

Remember that JSON isn't always the answer.
In your case, I would seriously consider a binary format.

Something like this:

```c++
client.publish("topic", &bankData , sizeof(bankData); 
```

If this format must evolve, you can either prefix it with a version identifier or use another topic.

If your not familiar with this technique, you can watch my video [Serialization and JSON on Arduino](https://www.youtube.com/watch?v=hAB4TdX8dwM); the first part talks about binary serialization.

Best regards,
Benoit

bobcroft, 2020-06-05:
Benoit, many thanks for your suggestion and wise advice.  I did not realize one could use client.publish etc. in the manner you show.  I'll try it in the next couple of days.  That will enable me to send the data as I wish and allow me refine my code as I had hoped to.  It feels like I have the best of both worlds now!
I have watched your video up to the point where you started to discuss JSON, I'll watch that later as I want to focus on the binary serialization initially.  The video is very good at explaining the concepts.
Thank you.

bobcroft, 2020-06-05:
Benoit, a quick question as my mind has gone blank!!  Doing as you suggest and send &bankData as the payload.  I subscribe to the ‘topic’ on the receiving device, how do I get the bankData in the payload into the bankData structure on the receiving device?  It’s probably straight forward but I am not seeing it.

 

Thanks

 

Bob

bblanchon, 2020-06-06:
Here is how you could copy the payload to `bankData`.

```
void callback(const char[] topic, byte* payload, unsigned int length)
{
   if (strcmp(topic, "topic") == 0 && length == sizeof(bankData)) {
     memcpy(&bankData, payload,  sizeof(bankData));
   }
}
```

bobcroft, 2020-06-06:
Hi Benoit,  Thank you so much for your help.  I had a quiet moment whilst taking a bath! And thought that perhaps memcpy would be the way to go.  Stay safe and I look forward to seeing more of your work.

 

Bob

bblanchon, 2020-06-07:
Alternatively, you can cast the `payload` pointer.

bobcroft, 2020-06-09:
Hi Benoit,

 

Once again thank you for all of your help.  I now how the transfer of my structure working.  I will share what I did and please do comment if you can see mistakes in what I do, or better ways as I am keen to learn.

 

To send the structure I use;

 

void sendStruct()

{

  uint16_t targetBank = (newData[0]) * 10;

 

  sprintf_P(stopic, PSTR("%s/%s/%i"), TPC_CMND, TPC_UPDATE, targetBank );

  mqttClient.publish(stopic, (byte*)&bankData , sizeof(bankData));

}

 

To receive the structure I use;

 

uint16_t receiveID = bankID * 10;     // bank indentifier

  sprintf_P(stopic, PSTR("%s/%s/%i"), TPC_CMND, TPC_UPDATE, receiveID);

  if (strcmp(topic, stopic) == 0 && data_len == sizeof(bankData))

  {

    memcpy(&bankData, data,  sizeof(bankData));

    updateEEprom();

 

In order for the code to work I had to add (byte*) to the publish command.  I think this what you meant in your last note.

I also had to change the size of the data_len in the mqtt call back from uint8_t to uint16_t because the structure size is 336 bytes.

I also changed the maximum buffer size in the PubSubclient library to accommodate the 336 bytes of data + topic.  In the latest version (May 2020) of the PubSubClient library this can be done with  mqttClient.setBufferSize(400); without having to change the entry in the core code.

 

I have written the code to that the settings for any ‘bank’ can be set on a control panel I have built (or using Blynk on a mobile phone) and then sent to the correct bank.  The system is a horticultural control and the ‘banks’ are banks of up to 8 relays or SSD’s that control water valves, pumps, motors etc. The system is flexible and can use interventions from soil moisture sensors, heat / humidity sensors, light sensors etc.  I tried to use the weather forecast to control irrigation but that wasn’t reliable enough.

 

All is looking good now. I have one final thought, I wish you were my tutor!!  At 67 years old learning is slow for me but I enjoy electronics, programming and now my new 3D printer.

 

Bob

closed 2020-06-16 by bblanchon