bblanchon/ArduinoJson #1288: Create json object from text file with ArduinoJson

dcota, 2020-06-07:
Dear Benoit,
I am facing some questions here when deciding the best way to do this. I have a text file in a SD Card. I need to read the file from SD and create a JSON object with that text and send it in a HTTP request to my database. 
I don’t really know what is the best strategy to use ArduinoJson to do this. I have tried to create a nested array and add while reading from SD but isn’t working.
Can you give me some clues?
Thanks again.
Duarte

bblanchon, 2020-06-07:
Hi Duarte,

Did you check [JsonConfigFile.ino](https://arduinojson.org/v6/example/config/)?

Best regards,
Benoit

dcota, 2020-06-07:
Hi Benoit, tnahk you.
Yes I tried to use that file but no success. I the meantime I realized that the problem is I dont have the file in a valid JSON format.
The file that I whant to serialize and send with ArduinoJson is this a function like this:
[code.txt](https://github.com/bblanchon/ArduinoJson/files/4743047/code.txt)
Any more clues?
Duarte

bblanchon, 2020-06-16:
Hi @dcota,

This looks like a source file.
Indeed, this has nothing to do with JSON.

Best regards,
Benoit

closed 2020-06-16 by bblanchon