bblanchon/ArduinoJson #1292: deserialization error!

samos2011, 2020-06-11:
Hi, I'm using V5.13.5, but I encounter a strange problem, with such a json data, deserialization always fails, but just adjust the format, reduce some indentation, delete an Array element, or delete a negative number `-`, or delete an Object, deserialization will always succeed.

What is the reason for this? I like this Lib, but is it reliable?
```json
{
  "ObjectList": [
    0,
    1,
    2,
    3,
    4,
    5,
    6,
    7,
    8,
    9,
    10,
    11,
    12,
    13,
    14,
    15,
    16,
    17,
    18,
    19,
    20,
    21,
    22,
    23,
    24,
    25,
    26,
    27,
    28,
    29,
    30,
    31
  ],
  "Start": 0,
  "TechHeadVer": 1,
  "StepVer": 1,
  "CycleCount": 1,
  "RecordTimeInterval": 10,
  "RecordVoltChange": 2,
  "RecordCurrChange": 1,
  "StepList": [
    {
      "Type": 4,
      "T": 2
    },
    {
      "Type": 11,
      "Val": 60000,
      "C1": 3600
    },
    {
      "Type": 12,
      "Val": -60000,
      "C1": 3000
    },
    {
      "Type": 1,
      "Val": 1,
      "C1": 10
    }
  ],
  "GlobalProtection": {}
}
```

Only the following code is used, the system heap is sufficient, `ARDUINOJSON_USE_LONG_LONG`, `ARDUINOJSON_USE_DOUBLE` is enabled
`len` is correct, and the terminator `\0`` is not missing
```c++
DynamicJsonBuffer jsonBuffer;
JsonObject& root = jsonBuffer.parseObject((const char*)buf, len);
if (!root.success()) {
	printf("parse failed!\n");
	return;
}
```

closed 2020-06-11 by samos2011

samos2011, 2020-06-11:
Sorry, the `parseObject` parameter is used incorrectly, without the `len` parameter

bblanchon, 2020-06-11:
The second parameter of [parseObject()](https://arduinojson.org/v5/api/jsonbuffer/parseobject/) is not the length but the nesting limit.