bblanchon/ArduinoJson #1294: print myObject without quotation marks

BogdanPau, 2020-06-12:
Hi guys,
I've been trying for a long time and online research for something that sounds quite basic.
I'm using this library in my Arduino project with ESP8266 NodeMCU LUA chip.
I'm fetching successfully the GET HTTP with an API I'm connecting to.
To API returning the JSON, example: 
`JSON object = {"meta":{"timestamp":"2020-06-12T19:46:07.708066Z","stations_updated":"2019-06-09"},"altimeter":{"repr":"Q1008","value":1008,"spoken":"one zero zero eight"},"clouds":[{"repr":"FEW034","type":"FEW","altitude":34,"modifier":null,"direction":null}],"flight_rules":"VFR","other":[],"sanitized":"LRCL 121930Z 31003KT 9999 FEW034 19/17 Q1008 R07/19//95","visibility":{"repr":"9999","value":9999,"spoken":"nine nine nine nine"},"wind_direction":{"repr":"310","value":310,"spoken":"three one zero"},"wind_gust":null,"wind_speed":{"repr":"03","value":3,"spoken":"three"},"wx_codes":[],"raw":"LRCL 121930Z 31003KT 9999 FEW034 19/17 Q1008 R07/19//95","station":"LRCL","time":{"repr":"121930Z","dt":"2020-06-12T19:30:00Z"},"remarks":"R07/19//95","dewpoint":{"repr":"17","value":17,"spoken":"one seven"},"remarks_info":{"dewpoint_decimal":null,"temperature_decimal":null},"runway_visibility":[],"temperature":{"repr":"19","value":19,"spoken":"one nine"},"wind_variable_direction":[],"units":{"altimeter":"hPa","altitude":"ft","temperature":"C","visibility":"m","wind_speed":"kt"}}
`

Then with the library parser I can take the objects I wish.
The problem is that when I print one of the objects out, let say:
`Serial.println(myObject["meta"]["timestamp"]);`
I get this result:
`Timestamp: "2020-06-12T19:46:07.708066Z"`
With the " " (quotation marks) and those are giving me an headache to get rid off in efficient way.
I know I can store the myObject in a string and then do String.replace("\"",""). But if I'm taking a lot of values from the JSON and clean them each time like this I'm just making super excessive code.
Is there any efficient way to clean the JSON from those quotation marks?
I acknowledge they might be from the returned API response, as it does has them..

Thanks,
Bogdan

bblanchon, 2020-06-13:
Hi Bogan,

I think the following line should work with both libraries:

```c++
Serial.println((const char*)doc["meta"]["timestamp"]);
```

See also:

* [Comparison between ArduinoJson and Arduino_JSON](https://arduinojson.org/news/2019/11/19/arduinojson-vs-arduino_json/)

Best regards,
Benoit

BogdanPau, 2020-06-13:
Hi Benoit,
Thanks for the help and for the helping comment!
When trying to work with your suggestion I got the following error:
`exit status 1
'doc' was not declared in this scope`

I saw also the compression of both libraries and it seems I should definitely use this ArduinoJson one. Perhaps your solution will work as well.
But to migrate from the two I need some help.
Can you tell me how may I parse the Json object using this library so I can migrate myself?

`JSONVar myObject = JSON.parse(jsonBuffer);
      // JSON.typeof(jsonVar) can be used to get the type of the var
      if (JSON.typeof(myObject) == "undefined") {
        Serial.println("Parsing input failed!");
        return;
      }
    
      Serial.print("JSON object = ");
      Serial.println(myObject);`

Then I will also try the 
`Serial.println((const char*)doc["meta"]["timestamp"]);`

Thanks,
Bogdan

BogdanPau, 2020-06-13:
My bad about the 
`exit status 1 'doc' was not declared in this scope`
Well I didn't see you are using doc as a variable.

I could make the migration and show the values without quotation marks as you showed using this library. 

Yet, I would like to configure doc as without giving it a size, as the GET from the API can give me different answers with more or less memory to use and I don't want to hard code the maximum memory allowance of the doc.
in Arduino_JSON the jsonBuffer was just a String and therefore I didn't needed to allocate the memory in advanced.

Any suggestions?

bblanchon, 2020-06-14:
Hi Bogdan,

Welcome to ArduinoJson! I highly recommend that you read the tutorials:

* [Deserialization tutorial](https://arduinojson.org/v6/doc/deserialization/)
* [Serialization tutorial](https://arduinojson.org/v6/doc/serialization/)

ArduinoJson allocates the memory pool before-hand; that's one of the secrets behind the performance, the reliability, and the code size.
It seems like a big deal at first, but really it's not; your MCU doesn't have infinite memory anyway.
Please see:

* [How to determine the capacity of the JsonDocument?](https://arduinojson.org/v6/how-to/determine-the-capacity-of-the-jsondocument/)

You're not supposed to keep your [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/) from one loop iteration to the other, but even if you do, you shouldn't need to call [`doc.clear()`](https://arduinojson.org/v6/api/jsondocument/clear/) because [`deserializeJson()`](https://arduinojson.org/v6/api/json/deserializejson/) does it for you.
Please see:

* [How to reuse a JsonDocument?](https://arduinojson.org/v6/how-to/reuse-a-json-document/)

Lastly, I don't see any variable declaration in the statement you mentioned. The only expensive operation in this expression is the lookup for the string "flight_rules" in the object.

Best regards,
Benoit

BogdanPau, 2020-06-14:
Hi,
Alright! Thanks again. I tried to read those articles, yet I'm not sure what I'm doing wrong that I'm not able to take out the clear() function without getting error of memory.

Also, did I understood right the difference between Dynamic doc to Static doc?
I did read on the documentation that static is faster and tends to smaller objects and dynamic is slower but tend to bigger size documents.
In my needed case, I am having a void loop that check a button state all the time,
If the button is pressed I'm fetching a new API call that needed to be stored for that loop only, print out the relevant data out of it and then I don't care if it will be cleared. thanks to your method I can print out without the quotation marks!
Yet, if when I didn't use the doc.clear() yesterday, I'm got an error of memory from the 2nd time pressing the button. like the code is trying to concatenate the old and the new data together.
Now I don't see that happening, so maybe it was a memory oversize fetching :(
What is the maximum doc size I can declare? I'm using now 4096. And is it right to declare it outside of the void loop? Or it just consuming more resources?
 
If you can, please take a look at my code and the output, I did also tried to print the JSON data size so you can have more information.
How should I use right way, without the clear then?

```
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
//#include <WiFiClient.h>
//#include <Arduino_JSON.h>
#include <ArduinoJson.h>

const char* ssid = "DIGI-02207951";
const char* password = "****";
const char* fingerPrint = "****";
String avwxApiKey = "****";

String serverPath;
String jsonBuffer;
DynamicJsonDocument doc(4096);

const int buttonPin = 4;
const int ledPin = 5;
int buttonState = 0; 
int oldbuttonState = 0;

void setup() {
  pinMode(buttonPin, INPUT);
  pinMode(ledPin, OUTPUT);
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  Serial.println("Connecting");
  while(WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to WiFi network "); Serial.print(ssid);  Serial.print(" with IP Address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  buttonState = digitalRead(buttonPin);
  if (oldbuttonState != buttonState && buttonState == HIGH) {
    digitalWrite(ledPin, HIGH); // fetching data led
    // Check WiFi connection status
    if(WiFi.status()== WL_CONNECTED){
      serverPath = "https://avwx.rest/api/metar/LRCL?token=" + avwxApiKey;
      jsonBuffer = httpGETRequest(serverPath.c_str());
      // Deserialize the JSON document
      //doc.clear(); // using clear not to override memory?
      DeserializationError error = deserializeJson(doc, jsonBuffer);
      // Test if parsing succeeds.
      if (error) {
        Serial.print(F("deserializeJson() failed: "));
        Serial.println(error.c_str());
        return;
      }

      Serial.print("doc Memory Usage = ");Serial.println(doc.memoryUsage());
      Serial.print("JSON object = ");
[... 46 lines cut ...]
```

And these are the Serial print lines:
(For now I'm printing the whole JSON object, just for debugging purposes)
```
Connected to WiFi network DIGI-02207951 with IP Address: 192.168.1.107
HTTP Response code: 200
doc Memory Usage = 2138
JSON object = {"meta":{"timestamp":"2020-06-14T16:10:09.098689Z","stations_updated":"2019-06-09"},"altimeter":{"repr":"Q1005","value":1005,"spoken":"one zero zero five"},"clouds":[{"repr":"SCT046","type":"SCT","altitude":46,"modifier":null,"direction":null},{"repr":"FEW050CB","type":"FEW","altitude":50,"modifier":"CB","direction":null}],"flight_rules":"VFR","other":[],"sanitized":"LRCL 141600Z 06004KT 040V100 9999 TS SCT046 FEW050CB 26/16 Q1005","visibility":{"repr":"9999","value":9999,"spoken":"nine nine nine nine"},"wind_direction":{"repr":"060","value":60,"spoken":"zero six zero"},"wind_gust":null,"wind_speed":{"repr":"04","value":4,"spoken":"four"},"wx_codes":[{"repr":"TS","value":"Thunderstorm"}],"raw":"LRCL 141600Z 06004KT 040V100 9999 TS SCT046 FEW050CB 26/16 Q1005","station":"LRCL","time":{"repr":"141600Z","dt":"2020-06-14T16:00:00Z"},"remarks":"","dewpoint":{"repr":"16","value":16,"spoken":"one six"},"remarks_info":{"dewpoint_decimal":null,"temperature_decimal":null},"runway_visibility":[],"temperature":{"repr":"26","value":26,"spoken":"two six"},"wind_variable_direction":[{"repr":"040","value":40,"spoken":"zero four zero"},{"repr":"100","value":100,"spoken":"one zero zero"}],"units":{"altimeter":"hPa","altitude":"ft","temperature":"C","visibility":"m","wind_speed":"kt"}}
Print Timestamp: 2020-06-14T16:10:09.098689Z
RAW REPORT: LRCL 141600Z 06004KT 040V100 9999 TS SCT046 FEW050CB 26/16 Q1005
VFR/IFR: VFR
W/V: 060/04ktHTTP Response code: 200
doc Memory Usage = 2198
JSON object = {"meta":{"timestamp":"2020-06-14T16:10:16.260789Z","stations_updated":"2019-06-09","cache-timestamp":"2020-06-14T16:10:09.081000Z"},"altimeter":{"repr":"Q1005","value":1005,"spoken":"one zero zero five"},"clouds":[{"repr":"SCT046","type":"SCT","altitude":46,"modifier":null,"direction":null},{"repr":"FEW050CB","type":"FEW","altitude":50,"modifier":"CB","direction":null}],"flight_rules":"VFR","other":[],"sanitized":"LRCL 141600Z 06004KT 040V100 9999 TS SCT046 FEW050CB 26/16 Q1005","visibility":{"repr":"9999","value":9999,"spoken":"nine nine nine nine"},"wind_direction":{"repr":"060","value":60,"spoken":"zero six zero"},"wind_gust":null,"wind_speed":{"repr":"04","value":4,"spoken":"four"},"wx_codes":[ [... cut ...] epr":"26","value":26,"spoken":"two six"},"wind_variable_direction":[{"repr":"040","value":40,"spoken":"zero four zero"},{"repr":"100","value":100,"spoken":"one zero zero"}],"units":{"altimeter":"hPa","altitude":"ft","temperature":"C","visibility":"m","wind_speed":"kt"}}
Print Timestamp: 2020-06-14T16:10:16.260789Z
RAW REPORT: LRCL 141600Z 06004KT 040V100 9999 TS SCT046 FEW050CB 26/16 Q1005
VFR/IFR: VFR
W/V: 060/04kt
```

Why the doc memory grew from 2138 to 2198? I have no idea..

Thanks,
Bogdan

ewaldc, 2020-06-15:
Continuously reusing the same JsonDocument isn't a recommended best practice as explained
[here](https://arduinojson.org/v6/how-to/reuse-a-json-document/).
Better to create a local `DynamicJsonDocument` at each fetch of the REST API, or return it from a function.  It's so fast, you won't even notice it...  One of my ESP apps generates on some days more than 100000 JsonDocuments/day without not a single byte leaking.

BogdanPau, 2020-06-15:
Hi @ewaldc 
Thanks for joining in here on my question.
So you suggest I should move the `DynamicJsonDocument doc(4096);` from the outside global deceleration into the void loop of Arduino?
Should I just place that DynamicJsonDocument statement or I should also make another command before/after?

Thanks,
Bogdan

bblanchon, 2020-06-16:
Hi @BogdanPau,

As a general design rule, I recommend avoiding global variables because they make the code harder to test and to reason about.
Additionally, in embedded programming, where resources are very limited, you cannot afford a 4KB buffer staying in memory when unused.
As @ewaldc, the best practice is to move the declaration of `doc` inside the loop, just before calling `deserializeJson()`.

Best regards,
Benoit

BogdanPau, 2020-06-16:
Great! I'll do so!

Thanks for your help, tips and tutoring in this case!

bblanchon, 2020-06-17:
You're welcome. Good luck with your project!

closed 2020-06-17 by bblanchon