bblanchon/ArduinoJson #1295: Does a non-blocking deserializeJSON(doc, stream) exist?

Nate711, 2020-06-14:
I have json data coming in over serial at a slower rate than my main control loop, which is running at 1kHz. Is there a way to iteratively build up a jsondocument between calls to the control loop? Thanks!

bblanchon, 2020-06-14:
Hi @Nate711,

Unfortunately, there is no such feature.
I understand why you would need this but there is no way I can implement it with the current design of the library.

As an alternative, I suggest that you accumulate the input in a buffer that you pass to `deserializeJson()` at the end.

Best regards,
Benoit

Nate711, 2020-06-16:
Thanks Benoit. I made a small accumulator and it's working with ArduinoJSON. I'm linking just for the googlers: https://github.com/Nate711/DJIPupperTests/blob/master/lib/NonBlockingSerialBuffer/NonBlockingSerialBuffer.h

closed 2020-06-16 by Nate711