bblanchon/ArduinoJson #1296: How to use a CustomReader

moser88704, 2020-06-17:
I don't know how to build a CustomReader for ESP8266. Following your example I have tried it as follows:

`
WiFiClient client;

struct myCustomReader {
    // Reads one byte, or returns -1
    int read() {
      int c = -1;
      if (client.available()) {
        c = client.read();
      }
      yield();
      return c;
    }

    // Reads several bytes, returns the number of bytes read.
    size_t readBytes(char* buffer, size_t length){
      size_t N = 0;
      int c;
      while ((client.connected() || client.available()) && (N < length)) {
        if (client.available()) {
          c = client.read();
          buffer[N] = c;
          N++;
        }
        yield();
      }
      return N;
    }
};
`
I have used code from a standard json streaming example:

`
myCustomReader refCustomReader;

DynamicJsonDocument doc(2048);

DeserializationError err = deserializeJson(doc, refCustomReader);
  if (err) {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(err.c_str());
  }
`
I receive the error: "InvalidInput"

The following code demonstrates that refCustomerReader works correctly:

`
  int c;
  char Buffer[128] = { 0 };
  myCustomReader refCustomReader;

  while (true) {
    c = refCustomReader.readBytes(Buffer, 128);
    if (c <= 0) {
      break;
    }
    for (int i = 0; i < c; i++) {
      Serial.printf("%c", Buffer[i]);
    }
  }
`

Thank you very much for your help.

Regards, Andi

bblanchon, 2020-06-17:
Hi Andi,

Fist of all, you don't need a custom reader for `WiFiClient` because [`deserializeJson()`](https://arduinojson.org/v6/api/json/deserializejson/) natively supports all standard stream classes.

Now, your implementation seems right, but the problem is that it stops reading as soon as `client.available()` returns false. In other words, it doesn't wait for the complete message, so the JSON document is likely to be truncated.

However, when the message is incomplete (or empty), [`deserializeJson()`](https://arduinojson.org/v6/api/json/deserializejson/) returns [`IncompleteInput`](https://arduinojson.org/v6/api/misc/deserializationerror/#incompleteinput), not [`InvalidInput`](https://arduinojson.org/v6/api/misc/deserializationerror/#invalidinput). There are other reasons that would explain [`InvalidInput`](https://arduinojson.org/v6/api/misc/deserializationerror/#invalidinput); the most obvious being an invalid input...

Did you check that the input document is valid?
More importantly, why did you create a custom reader?

Best regards,
Benoit

moser88704, 2020-06-17:
Hi Benoit

Thank you very much for your answer and your suggestions. i would like to address them individually:

**Use WiFiClient:** Indeed I have first tried to directly use the WiFiClient, which also resulted in a "InvalidInput" error. This is the reason for trying it with the CustomReader.

**Invalid Input:** Interestingly, the following code works (even though very slowly, since the document is large):
```cpp
  String result;
  while(client.available()) {
      result = client.readStringUntil('\r');
  }
  DeserializationError err = deserializeJson(doc, result);
  if (err) {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(err.c_str());
  }
  Serial.println("serializeJsonPretty");
```
So I assume that the input document is correct. 

**Potential custom reader problem:**: For debugging purposes I have added a "Serial.printf("1");" at the beginning of both calls. I do not get any output in the monitor window from the print command. This makes me believe that none of the functions gets called.

As mentioned at the end of my first post, myCustomReader seems to work properly and does not have problems with client.available(), etc. The output from the last script in my first post can be correctly interpreted by Firefox.

**Additional infos:** I am trying to load the document from openwheather.org. I use the following code to connect to the www-site:
```cpp
  String servername = "api.openweathermap.org";  // remote server we will connect to
  String APIKEY = "xyz";
  const int httpPort = 80;
  if (!client.connect(servername, httpPort)) {
        return;
    }
      // We now create a URI for the request
  String url = "/data/2.5/onecall?lat=37.233340&lon=-121.801760&exclude=minutely,daily&appid="+APIKEY;

      // This will send the request to the server
  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
                "Host: " + servername + "\r\n" +
                "Connection: close\r\n\r\n");

```

This code works for all other cases except when I try to stream.

Thanks again for your help.

Regards, Andi

bblanchon, 2020-06-18:
Hi Andi,

I have no reason to believe that the custom reader feature is buggy: it's covered by the unit tests and many users are using it without any problem.

In the case studies of [Mastering ArduinoJson](https://arduinojson.org/book/), I demonstrate how to download forecast information from OpenWeatherMap. I use `WiFiClient` as well; it's for another architecture but the interface is identical.

However, in the case study, I make sure:

1. to avoid `String`s like the plague because they ruin the heap
2. to use `HTTP/1.0` to prevent [Chunked Transfer Encoding](https://en.wikipedia.org/wiki/Chunked_transfer_encoding)
3. to filter the input and ignore all irrelevant fields

Best regards,
Benoit

moser88704, 2020-06-20:
Hi Benoit

The custom reader works with the following read() function:

```ccp

  bool isBody = false;

  int read()
  {
    int c = -1;
    {
      while (!isBody)
      {
        c = client.read();
        if (c == '{' || c == '[')
        {
          isBody = true;
          return c;
        }
      }
      while (c == -1)
      {
        yield();
        c = client.read();
      }
    }
    yield();
    return c;
  }
```

With this code I handle the problem you have mentioned in bullet 2).

Thank you very much for your help.

Regards, Andi

closed 2020-06-20 by moser88704

bblanchon, 2020-06-20:
Hi Andi,

I'm pretty sure this implementation fails if the body is made of multiple chunks.
Why don't you use HTTP/1.0?

Best regards,
Benoit