bblanchon/ArduinoJson #1297: _DoubleExceptionVector crash

Tintin4000, 2020-06-17:
I am using the Azure IoT Library (1.3.8) on an ESP32 (1.0.4), upon reception of a TWINCallback, I am trying to deserialize the JSON payload.
At the creation of the JsonDocument (either Static or Dynamic) a core crash is observed.
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: BREAK instr

Using VSMicro and GDB along with a ESP-PROG, I can trace the halt down to the following module:
Breakpoint 2, updateDeviceTwinCallback (updateState=<optimized out>, payLoad=<optimized out>, size=<optimized out>, userContextCallback=<optimized out>) at C:\Users\didie\Documents\Arduino\Azure-IoT-Garden\Hub\ESP32_Garden_Hub/ESP32_Garden_Hub.ino:743
743		DynamicJsonDocument jsonTwinDoc(MAX_JSON_TWIN_SIZE);
@"Target halted. PRO_CPU: PC=0x4011E40E             APP_CPU: PC=0x400803C0 (active)\n"
Program received signal SIGTRAP, Trace/breakpoint trap.
_DoubleExceptionVector () at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:543
Trace/breakpoint trap.

Any help would be appreciated ;-)

Tintin4000, 2020-06-18:
I have duplicated the same behavior on a Adafruit Feather ESP32 and the Espressif ESP32 DevKitC.

bblanchon, 2020-06-18:
Hi @Tintin4000,

I'm sorry but I don't know how to solve this error.
At first, I thought it could be a problem with `malloc()` or a stack overflow but you specifically said that both `DynamicJsonDocument` and `StaticJsonDocument` are affected.

I think that's a question for the ESP32 or Azure IoT Library teams; however, if you can write an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example) we can try to reproduce the issue too.

Best regards,
Benoit

Tintin4000, 2020-06-18:
The code of the project can be found at https://github.com/Tintin4000/Azure-IoT-Garden-Irrigation/blob/Update-Sensor-Communication/Hub/ESP32_Garden_Hub/ESP32_Garden_Hub.ino.
I am not sure that I can strip it down, because then the error disappears as well as the project process flow and purpose.  The issue is in the updateDeviceTwinCallback callback function upon initialization of the JsonDoc.  If you see anything you would not recommend in the code please let me know.  I am using the two cores so maybe something around that?  I will start all the tasks under the same core and see if this has any effects?
Didier

Tintin4000, 2020-06-18:
Same behavior with all the tasks running in a single core.

bblanchon, 2020-06-18:
I really doubt the crash occurs in the constructor of `StaticJsonDocument`.
Can you insert more traces to make sure this is really the line of the crash?

I don't think you need to allocate the `temp` buffer.
Why don't you pass `payload` directly to [`deserializeJson()`](https://arduinojson.org/v6/api/json/deserializejson/)?

Tintin4000, 2020-06-18:
It is this line of code specifically that cause the crash (obviously could be related to something else).  When I rem all the code related to parsing the JSON in the callback the code run to the end.  payload is read only so I thought that JSON would then not be able to modified it instead of temp and save some memory?

Tintin4000, 2020-06-18:
I will try to create a queue to process the TWIN message in a task and return from the callback rather faster.  Will keep you posted.

Tintin4000, 2020-06-18:
After moving the processing of the DeviceTWIN callback to a task invoking the JSONDoc contructor, I have no crash anymore.
It isn't could practice to process data in callback function anyway, so I will leave it like this.  Thank you so much for looking into it.
It will stay a mystery ;-) for now.

bblanchon, 2020-06-18:
I'm glad you found a solution.

Is it possible that the function was running in the context of an interruption handler?
Maybe the word "vector" in `xtensa_vectors.S`  meant "interrupt vector"...

Anyway, I don't understand why the constructor of `StaticJsonDocument` would crash in an interrupt handler.
Sure, it could cause a stack overflow, but why would we get the same error with a `DynamicJsonDocument`...

Tintin4000, 2020-06-18:
I don't believe so, I am not running any interrupt handler in this project.  My best guess would be some unexpected interaction between FreeRTOS, the Azure client and some obscure unmanaged memory allocation.
I am happy to deploy my sensor and hub since the weather is definitely inviting for some irrigation in my front garden ;-).