bblanchon/ArduinoJson #1298: filtering for deserializeMsgPack()

imwhocodes, 2020-06-22:
Hi

Reading the documentation for[ ArduinoJson V6.15.0](https://arduinojson.org/news/2020/03/22/version-6-15-0/) seem like currently there is not support for the filtering option on `deserializeMsgPack()`

Is there any plan to add it in a next release? Or is a minor feature not in pipeline right now?

Thanks,
Luca

bblanchon, 2020-06-22:
Hi Luca,

It's quite a lot of work, but that's something I can do if users are interested.
The problem with MessagePack is that very few of ArduinoJson users actually use it.
Let's see how many thumbs up you get.

Best regards,
Benoit

imwhocodes, 2020-10-02:
I'm reading the source code trying to understand the viability of such enhancement and if I can be of any help

From what I got confronting `JsonDeserializer.hpp` and `MsgPackDeserializer.hpp` in the latter the `skipObject()`, `skipArray()` ecc are missing, such functions keep parsing recursively the data without allocating on the memory

And the filter is a JsonDocument like traversed at the same time of the parsing and:

> a parse call can result in a parse call or skip call
> a skip call can only result in a skip call

Or I'm missing something?

imwhocodes, 2020-10-02:
I started working on it, going to continue next days
(Implemented skip/consumer, now remaining filtering decision)
https://github.com/imwhocodes/ArduinoJson/commit/5178a7964ec2635e3494344113142e241e670046

bblanchon, 2020-10-03:
I didn't realize you needed this feature so bad.
I'll start working on it and try to ship it in the next release.

imwhocodes, 2020-10-03:
Actually I already implemented it,
I made a PR https://github.com/bblanchon/ArduinoJson/pull/1394

The code is completely based on deserializeJson one
I ran the included unit testing and everything passed

I tried to convert Json filtering unit test to msgpack but in that test there are a lot of cases that are meaningless on msgpack so given that the PR do not have any regression or breaking change I'm going to leave them for later

closed 2020-10-13 by bblanchon

bblanchon, 2020-10-19:
Filtering for MessagePack is now available in [ArduinoJson 6.17.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.17.0).

commit 71200521 (2020-10-13): Added filtering for MessagePack (closes #1298, closes #1394)