bblanchon/ArduinoJson #1299: DeSerialization of multiple json documents from one file with a terminating char! ( ESP32 Arduino )

zekageri, 2020-06-22:
### I have a file wich contains multiple json documents separated by '\n' or any special terminating character.

**The problem i'am facing is that the error this code is returning is "IncompleteInput".
Hovewer the chars are good, i can see it inside the FILE and i debugged it while serialize with <StreamUtils.h>.**

_I really need to store these json objects inside one file and separate them by a terminating character. In this case '\n'._

### Here is the code:

```
static const inline void Get_User_Prog(){
    const size_t capacity = 15000;
    File JSONFILE = SPIFFS.open("/UserProg.json", "r");
    if(!JSONFILE){
        Error_iras("Can't open UserProg.json for read.");
    }else{
        DynamicJsonDocument doc(capacity);
        do{
            DeserializationError error = deserializeJson(doc, JSONFILE);
            if (error) {
                String Error = "JSOn Error ~ ";Error += String(error.c_str());
                Error_iras(Error);
            }else {
                JsonArray content = doc["content"];
                JsonArray content_0_content = content[0]["content"];
                for(int x = 0; x < 16;x++){
                    JsonObject content_0_content_0_attributes = content_0_content[x]["attributes"];
                    const char* Addr    = content_0_content_0_attributes["outexp"];
                    const char* pinke   = content_0_content_0_attributes["outpin"];
                    if(pinke[0] != 0){
                        int AddressIndex = String(Addr).indexOf(":");
                        int Address = (String(Addr).substring(AddressIndex+1)).toInt();
                        int PIN     = (String(pinke).substring(7)).toInt();
                        Timers[PIN].Address = Address;
                    }
                }
            }
        }while(JSONFILE.read() == '\n');
    }
```
    JSONFILE.close();
}

bblanchon, 2020-06-23:
Hi @zekageri,

As I told you yesterday on the chat, [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/) returns [`IncompleteInput`](https://arduinojson.org/v6/api/misc/deserializationerror/#incompleteinput) when the input is empty.
I wish it could return a dedicated code, and maybe that's something I'll add in the future.

In your case, however, you don't need a dedicated error code, you simply need to check `JSONFILE.available()`.

BTW, if you use a `\n` as the separation character, you don't need to manually skip it: [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/) already skips spaces at the beginning of the input.

In summary:

```c++
while (JSONFILE.available()) {
  DeserializationError error = deserializeJson(doc, JSONFILE);
  // ...
}
```

Note that this technique is presented in the "Advanced Techniques / JSON Streaming" section of [Mastering ArduinoJson](https://arduinojson.org/book/).

Best regards,
Benoit

zekageri, 2020-06-23:
Thank you! Sorry for my little knowledge about ArduinoJson, it is very complex and require so much time to read and try it all. 
In the future I will master it based on the tutorials.

closed 2020-06-23 by zekageri

bblanchon, 2020-06-23:
Honestly, you spent more time with me than you'd need to read the tutorials.