bblanchon/ArduinoJson #1300: Serialize-deserialize JSON to Same board

regina2898, 2020-06-24:
Hi, can anyone help me?
I have this arduino code to serialize a JSON  : 
{
 session id = .... (filled in nodered),
 idmeter = [...] (filled in nodered)
}
to my node-red dashboard.

```
`#include <ArduinoJson.h>

void setup() {
  // Initialize Serial port
  Serial.begin(9600);
  while (!Serial) continue;
  StaticJsonDocument<200> doc;
  
  // Sessionnid name with null value
  JsonObject sessionid = doc.createNestedObject("sessionid");
  doc["sessionid"] = (char*)0; // or (char*)NULL if you prefer

  // Add an idmeter array.
  JsonArray idmeter = doc.createNestedArray("idmeter");

  // Generate the minified JSON and send it to the Serial port.
  //
  //serializeJson(doc, Serial);
  // to print :
  // {"sessionid":" ","idmeter":[ ]}

  // Start a new line
  Serial.println();

  // Generate the prettified JSON and send it to the Serial port.
  serializeJsonPretty(doc, Serial);
    // The above line prints:
  // {
  //   "sessionid": " ",
  //   "idmeter": "[ ]"
  // }
}

```
void loop() {
  // not used in this example
}`

And here's my Node-Red code 
`[{"id":"81e6b612.bd4468","type":"debug","z":"ff0d843c.5e05b8","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"jsonata","x":700,"y":240,"wires":[]},{"id":"86336189.e237d","type":"serial in","z":"ff0d843c.5e05b8","name":"","serial":"4bff77f8.029b88","x":90,"y":220,"wires":[["31fe5d6.317b0a2"]]},{"id":"31fe5d6.317b0a2","type":"json","z":"ff0d843c.5e05b8","name":"","property":"JSONata","action":"obj","pretty":true,"x":250,"y":220,"wires":[["f514a3f0.6b85d","25087f18.702cb"]]},{"id":"f514a3f0.6b85d","type":"ui_text_input","z":"ff0d843c.5e05b8","name":"","label":"","tooltip":"","group":"119a7de4.6ebb92","order":1,"width":0,"height":0,"passthru":true,"mode":"text","delay":"3000","topic":"","x":400,"y":280,"wires":[["21b8eff3.8f3e7"]]},{"id":"21b8eff3.8f3e7","type":"json","z":"ff0d843c.5e05b8","name":"","property":"payload","action":"obj","pretty":false,"x":550,"y":240,"wires":[["81e6b612.bd4468"]]},{"id":"25087f18.702cb","type":"ui_text","z":"ff0d843c.5e05b8","group":"119a7de4.6ebb92","order":1,"width":0,"height":0,"name":"","label":"","format":"{{msg.payload}}","layout":"row-left","x":390,"y":160,"wires":[]},{"id":"4bff77f8.029b88","type":"serial-port","z":"","serialport":"COM3","serialbaud":"19200","databits":"8","parity":"none","stopbits":"1","waitfor":"","newline":"1000","bin":"false","out":"interbyte","addchar":"false","responsetimeout":""},{"id":"119a7de4.6ebb92","type":"ui_group","z":"","name":"Received from Arduino","tab":"e2d4c61b.53b908","order":1,"disp":true,"width":"10","collapse":false},{"id":"e2d4c61b.53b908","type":"ui_tab","z":"","name":"Home","icon":"dashboard","disabled":false,"hidden":false}]`

[image]

I already can fill the missing value from node red. But I'm having problem to deserialize it to my Arduino (I wish to deserialize the filled JSON to the same board that I used to serialize the blank JSON)

Is it possible for me to do that? 

Thank you!

regina2898, 2020-07-06:
@bblanchon 
Thank you, Benoit

In software serial communcation, can a sender be a receiver to at the same time (and vice versa)?

bblanchon, 2020-07-08:
Yes, you can send a receive from the same board, provided that you wired the TX and RX pins (and ground!) correctly.

closed 2020-07-17 by bblanchon