bblanchon/ArduinoJson #1302: Can I send a string\serialize in a POST or do I need to convert somehow?

Kodekat84, 2020-06-26:
esp32
arduino 1.8.12

I am GET'ing a json which I am saving over to globals so I can then use some of the data in a post:
`   const size_t capacity = JSON_OBJECT_SIZE(2) + JSON_OBJECT_SIZE(6) + 200;
      DynamicJsonDocument doc(capacity);

      String pollingPayload = http.getString();

      const char* json = pollingPayload.c_str();
      //deserializeJson(doc, json);
      //const char* json = pollingPayload.toString();
      //deserializeJson(doc, (const char*)json);
      deserializeJson(doc, (const std::string&)json);

      int status = doc["status"]; // 0

      JsonObject payload = doc["payload"];
      //const char* payload_id = payload["id"]; // "2e939706-8434-407d-9463-ea680b1a5231"
      //const char* payload_lanyardId = payload["lanyardId"]; // "987654444"
      //const char* payload_scanned = payload["scanned"]; // "2020-06-20T08:43:08.000+0000"

      String payload_id = payload["id"]; // "2e939706-8434-407d-9463-ea680b1a5231"
      const char* payload_activityId = payload["activityId"]; // "c528c641-50a3-498b-9f6a-37342c58f195"
      String payload_lanyardId = payload["lanyardId"]; // "987654444"
      String payload_scanned = payload["scanned"]; // "2020-06-20T08:43:08.000+0000"
      int payload_state = payload["state"]; // 1
      int payload_success = payload["success"]; // 0

      //Copy the stuff over to globals
      payloadIN_id = payload_id;
      payloadIN_lanyardId = payload_lanyardId;
      payloadIN_scanned = payload_scanned;
      payloadIN_state = payload_state;
      payloadIN_success = payload_success;

    }

    else {
      Serial.println("Error on HTTP request");
    }

    http.end(); //Free the resources

  }
  digitalWrite(25, LOW);
}`

Then I am trying to use these globals to send some over the same data back to the DB anlong with other game data:
` const size_t capacity = JSON_ARRAY_SIZE(3) + 3 * JSON_OBJECT_SIZE(5) + JSON_OBJECT_SIZE(9);
  DynamicJsonDocument docOut(capacity);

  docOut["id"] = payloadIN_id;//"811a885b-4bb9-410b-ab4f-dcf45fd3c76b";
  docOut["activityId"] = "c528c641-50a3-498b-9f6a-37342c58f195";
  docOut["lanyardId"] = payloadIN_lanyardId;//"B212EF3F8A1B412A94F3709A8A8AFE96";
  docOut["scanned"] = "2020-05-20T10:33:23.223";                //payloadIN_scanned; extra+0000
  docOut["startTime"] = "2020-05-20T10:36:42.002";
  docOut["endTime"] = "2020-05-20T10:47:47.885";
  docOut["state"] = 4;
  docOut["accuracy"] = 2;                                       // TODO add total completions

  JsonArray payload = docOut.createNestedArray("payload");

  JsonObject payload_0 = payload.createNestedObject();
  payload_0["id"] = payloadIN_id;//"811a885b-4bb9-410b-ab4f-dcf45fd3c76b";
  payload_0["playerActivityId"] = activityId;//"c528c641-50a3-498b-9f6a-37342c58f195";
  payload_0["time"] = "2020-05-20T10:42:22.112";                            //may need to be "2020-05-20T10:47:25.000+0000"
  payload_0["target"] = 9;
  payload_0["success"] = 0;

  JsonObject payload_1 = payload.createNestedObject();
  payload_1["id"] = payloadIN_id;//"811a885b-4bb9-410b-ab4f-dcf45fd3c76b";
  payload_1["playerActivityId"] = activityId;//"c528c641-50a3-498b-9f6a-37342c58f195";
  payload_1["time"] = "2020-05-20T10:46:55.654";
  payload_1["target"] = 8;
  payload_1["success"] = 1;

  JsonObject payload_2 = payload.createNestedObject();
  payload_2["id"] = payloadIN_id;//"811a885b-4bb9-410b-ab4f-dcf45fd3c76b";
  payload_2["playerActivityId"] = activityId;//"c528c641-50a3-498b-9f6a-37342c58f195";
  payload_2["time"] = "2020-05-20T10:47:25.685";
  payload_2["target"] = 7;
  payload_2["success"] = 1;

  String jsonOut;
  serializeJson(docOut, jsonOut);
  //Serial.println(String(baseURL) + String(postPlayerActivityData));
  serializeJsonPretty(docOut, Serial);
`

It sends, but where I have used the globals instead of the dummy data (same length) my JSON is truncated as if there are hidden characters or something I'm not seeing.  Here is the output from the serializeJsonPretty:
{
18:01:55.895 ->   "id": "07d697c4-3976-4a5e-9b55-85db2a1fe384",
18:01:55.895 ->   "activityId": "c528c641-50a3-498b-9f6a-37342c58f195",
18:01:55.895 ->   "lanyardId": "999999999",
18:01:55.895 ->   "scanned": "2020-05-20T10:33:23.223",
18:01:55.895 ->   "startTime": "2020-05-20T10:36:42.002",
18:01:55.895 ->   "endTime": "2020-05-20T10:47:47.885",
18:01:55.895 ->   "state": 4,
18:01:55.895 ->   "accuracy": 2,
18:01:55.895 ->   "payload": [
18:01:55.895 ->     {
18:01:55.895 ->       "id": "07d697c4-3976-4a5e-9b55-85db2a1fe384",
18:01:55.895 ->       "playerActivityId": "c528c641-50a3-498b-9f6a-37342c58f195",
18:01:55.895 ->       "time": "2020-05-20T10:42:22.112",
18:01:55.895 ->       "target": 9,
18:01:55.943 ->       "success": 0
18:01:55.943 ->     },
18:01:55.943 ->     {
18:01:55.943 ->       "id": "07d697c4-3976-4a5e-9b55-85db2a1fe384",
18:01:55.943 ->       "playerActivityId": "c528c641-50a3-498b-9f6a-37342c58f195",
18:01:55.943 ->       "time": "2020-05-20T10:46:55.654"
18:01:55.943 ->     }
18:01:55.943 ->   ]

which is missing all the info for both target 7 and 8.  ALL of the correct data is sent if I remove the globals and stick to dummy data, i.e. docOut["lanyardId"] = payloadIN_lanyardId;//**"B212EF3F8A1B412A94F3709A8A8AFE96";**

What am I missing? do I need to convert the strings back to chars and then POST?

bblanchon, 2020-06-26:
Hi @Kodekat84,

Please do the effort of cleaning and formatting your message, I don't have time to decipher this 100+ lines mess.

To answer the question in the title: yes you can POST a JSON document; simply read the [Serialization Tutorial](https://arduinojson.org/v6/doc/serialization/) to the end.

Best regards,
Benoit

Kodekat84, 2020-06-26:
Thank you and sorry for the muddled code segments.

In my initial GET request function I assign an incoming payload to a string (payload_id):

```
String pollingPayload = http.getString();

const char* json = pollingPayload.c_str();

deserializeJson(doc, (const std::string&)json);

String payload_id = payload["id"];                        //expected data "2e939706-8434-407d-9463-ea680b1a5231"

```
 I then copy this over into a global so I can utilize it in another function down the line(payloadIN_id):

```
 payloadIN_id = payload_id;
```

Afterwards, in my POST function I pass (payloadIN_id) into the new JSON doc name docOut before serializing:

```
 docOut["id"] = payloadIN_id;                               //"811a885b-4bb9-410b-ab4f-dcf45fd3c76b";
```

 Yet this breaks the formatted json.  When I output to serial many of the final elements are now missing.  If I replace the above line with the same hard coded data, it works as expected.  Is this buffer overflow as I've been reading about?

Thanks again

Kodekat84, 2020-06-26:
Looks like I have it working utilizing c_str();   

like so:

`docOut["id"] = payloadIN_id.c_str();`

Is there a more efficient solution than me saving to a global String,just to convert to a string before sending out the JSON?

ewaldc, 2020-06-27:
The best way to deserialize from HttpClient, is to let ArduinoJson read directly from the network stream and bypass the use of Strings all together. See example [here](https://arduinojson.org/v6/example/http-client/).

`DeserializationError error = deserializeJson(doc, client);`

Alternatively you can deserialize directly from members of the String class per the [documentation](https://arduinojson.org/v6/api/json/deserializejson/)
`deserializeJson(doc, pollingPayload);`

The latter is less efficient though due to the way the String class is implemented.

For getting data out of a Json Document/Variant, a `const char *` is the most efficient as no data will get copied (basically you get a pointer to the data stored in the JsonDocument).
`const char * payload_id = payload["id"]`
When using a String, the String class will make a copy, since ArduinoJson internally does not use Strings.

When storing data into JsonDocument, it will depend on whether the data that you want to store will be kept in scope/memory during the lifetime of the JsonDocument.
E.g. for global data, data in Flash or data that will stay alive long enough, use or typecast to a writable type e.g. _char *_  so that ArduinoJson uses zero-copy mode as explained [here](https://arduinojson.org/v6/api/json/deserializejson/).
For non-writable types e.g. _const char *_ ArduinoJson will make a copy (for which no need enough capacity in the JsonDocument), in which case you do not have to worry about the original data which can e.g. be released.

closed 2020-07-17 by bblanchon