bblanchon/ArduinoJson #1304: Using ArduinoJSON with arm-gcc, libopenCM3 and unity/ceedling/cmock

krish2487, 2020-06-27:
Hello,
I apologize if it seems like a common or stupid question. I am trying to use a ArduinoJSON in a C project.
A little bit of background information - I am working with the arm-none-eabi-gcc compiler, libopencm3 HAL for the STM32 microcontroller and ArduinoJSON library and makefiles.
I am also using ceedling/unity/cmock framework to test the C code ( This is important as I cannot migrate to using a C++ compiler, I would like the code to be in C since the limiting factor is the test framework).
I would like to use the ArduinoJSON library (which is a C++ library) to create / parse some JSON structures.

My data structure file looks like so
```
#ifndef DATASTRUCTURES_H
#define DATASTRUCTURES_H

#include "exit_codes.h"
#include "ArduinoJson/ArduinoJson.h"
const char* template_POSTResults;
const char* template_RuntimeSettings;
const char* template_SensorPayload;
const char* template_GSMPayload;

EXIT_STATUS InitJSON(const char* template, DynamicJsonBuffer& Buffer);
EXIT_STATUS SetJSONValue(void);

#endif // DATASTRUCTURES_H

```
Basically in the InitJSON function I would like to take a template to a datastructure, take a reference to a DynamicJsonBuffer, set all the key values to NULL and set it to a known state. However, I am at a loss as to how to do it.

Trying to compile the code with the basic boilerplate gives the following result ( and understandably so).

```
  CC    main.c
  CC    appLayer.c
In file included from ../libs/ArduinoJson/ArduinoJson.h:5,
                 from DataStructures.h:6,
                 from appLayer.c:3:
../libs/ArduinoJson/src/ArduinoJson.h:15:2: error: #error ArduinoJson requires a C++ compiler, please change file extension to .cc or .cpp
 #error ArduinoJson requires a C++ compiler, please change file extension to .cc or .cpp
  ^~~~~
In file included from appLayer.c:3:
DataStructures.h:14:44: error: unknown type name 'DynamicJsonBuffer'
 EXIT_STATUS InitJSON(const char* template, DynamicJsonBuffer& Buffer);

```
I am no great shakes at C++ ( For that matter I am not great shakes at C), but I understand that the traditional way would be to tinker with the C++ library to make it compatible with C compilers. However I would prefer to avoid such an approach. I would like to keep the github link to the library untouched as it is likely to be used as a git submodule in the project. Hence no changes to the library if possible.
I would be glad and thankful if someone more experienced can guide me how to use a C++ library in a C source.

I am looking for some help / guidance on how to create a C interface to the ArduinoJSON library that I can then use in the final code and mock it out for testing while use the underlying library in production code.

Thanks in advance.

bblanchon, 2020-06-27:
Hi @krish2487,

The only way to use a C++ library in a C project is to wrap it with a C interface, but you still need a C++ compiler to build the module.
However, in the case of ArduinoJson, this wrapper would contain many (probably a hundred) functions because the library uses a lot of templates to adapt to the types used in the program.
I don't think it makes sense to do that.

There are many good C libraries for JSON, check-out cJSON or jsmn, for example.

Best regards,
Benoit

krish2487, 2020-06-28:
Hello @bblanchon ,
Thank you for the quick response. :-)
My apologies in the delay getting back to you. I understand that the cons outweigh the pros in this case while trying to create such an interface. And you are right, I think I will stick with a C library for now.  I am closing the issue now.

PS: Just as an additional comment, ArduinoJSON does bring the best features of many such json libraries into one umbrella. Thank you for an awesome library!! :-)

closed 2020-06-28 by krish2487