bblanchon/ArduinoJson #1305: Deserialization

gent99, 2020-06-29:
How to catch an exception when there's no match? I'having the following code:

  char CardBuffer[10];
  sprintf(CardBuffer, "%lu", cardId);

  UserID = doc[CardBuffer];

If my doc contain an instance for CardBuffer everything is fine. If not, my ESP8266 reboots with Exception (28). How can i catch an exception to prevent rebooting?

ewaldc, 2020-06-30:
Read [this ](https://arduinojson.org/v6/api/jsonobject/containskey/) section of the documentation.
The ESP crashes probably because you try to print/read from a null pointer.

bblanchon, 2020-06-30:
Alternatively, you can use the ["or" operator](https://arduinojson.org/v6/api/jsonvariant/or/) to provide a default value:

```c++
UserID = doc[CardBuffer] | "anonymous-user";
```

gent99, 2020-06-30:
Thank you. I refined my code and now it's not crashing anymore.

closed 2020-06-30 by gent99