bblanchon/ArduinoJson #1306: The elements in the JsonArray getting replaced when I use JsonArray::add().

Bharath-M-R, 2020-06-30:
Hi,
For the first time ,when data received its getting stored into the jsonArray properly.But if the next time data comes,I expect the data to be added at the end of the current jsonArray,But what happening is that the newly came data replaces the first came and getting added at the last also.

This is that part of code:

StaticJsonBuffer <2048>jsonBuffer1;
StaticJsonBuffer <2048>jsonBuffer2;

JsonObject& wifilist = jsonBuffer2.createObject();
JsonArray& mac_list = wifilist.createNestedArray("MAC");
JsonArray& rssi_list = wifilist.createNestedArray("RSSI");
JsonArray& last_seen = wifilist.createNestedArray("LAST_SEEN");

String payload = Serial.readString();

JsonObject& parsed = jsonBuffer1.parseObject(payload);

void loop()
{
  while(Serial.available())
  {
     for(int i=0;i< parsed_contain_array_length ; i++)
     {
         mac_list.add(parsed["MAC"][i]);
     }

Any solutions...?

Thanks,
Bharath

bblanchon, 2020-06-30:
Hi Bharath,

These `StaticJsonBuffer`s are huge!
What platform are you using?

Also, this program is very curious; can you explain the logic?

Note that ArduinoJson 5, stores `JsonObject` by reference.
When you do `mac_list.add(parsed["MAC"][i])`, you add a pointer from `jsonBuffer2` to `jsonBuffer1`.
So if you modify the object in one document, it will appear as changed in the other because it's the same instance.
This is different from ArduinoJson 6, which copies the object.

Best regards,
Benoit

Bharath-M-R, 2020-07-01:
Hi,
I will explain the situation.
I have two modules, every one minute the first module sends the json data to the second module through serial connection.
In the second module I need to receive, parse it and store it into another jsonArray and whenever data receives I want to add these data into the JsonArray . the JsonArray keep on getting added by the data sent by first module.
I hope u understand my explanation.
thanks

Bharath-M-R, 2020-07-01:
Hi bblanchon,
 Thanks for your response..  :)
My issue resolved by using only one jsonBuffer instead of the two. Thank you.

closed 2020-07-01 by Bharath-M-R