bblanchon/ArduinoJson #1307: Node lost after a while

guestisp, 2020-07-01:
Hi,
i'm getting some strange issues when using ArduinoJson in a loop.
This is a very simple code (not exactly what i'm using, for brevity)

```
const int capacity = JSON_ARRAY_SIZE(8) + 8 * JSON_OBJECT_SIZE(2);
DynamicJsonDocument json_mqtt_state(10240);

void readTemperatures() {
  TempSensors.begin();
  numberOfDevices = TempSensors.getDeviceCount();

  if ( numberOfDevices < 1 )
    return;

  TempSensors.requestTemperatures(); // Send the command to get temperatures
  
  StaticJsonDocument<capacity> jsonTemperatures;
  String jsonOutput;

  for(int i=0;i<numberOfDevices; i++){
    if ( TempSensors.getAddress(tempDeviceAddress, i) ) {
      float tempC = TempSensors.getTempC(tempDeviceAddress);
      String hexAddressString = hexAddress(tempDeviceAddress);

      jsonTemperatures[i]["probe"] = hexAddressString;
      jsonTemperatures[i]["temperature"] = tempC;
    }
  }

  json_mqtt_state["temperatures"] = jsonTemperatures;

  serializeJson(jsonTemperatures, jsonOutput);

  if ( jsonOutput != "null" )
    Serial.println(jsonOutput);
}

void sendStatusUpdate() {
  if ( !mqttClient.connected() ) {
    return;
  }

  json_mqtt_state["node"] = mqtt_client_id;
  json_mqtt_state["key"]["value"] = "Hello World!";

  String mqtt_payload = "";
  serializeJson(json_mqtt_state, mqtt_payload);

  Serial.println("Status JSON: " + mqtt_payload);
}

Both functions are called with TaskScheduler
What i'm seeing is that `mqtt_payload` is created properly from the start, with the node `temperatures` properly filled with the `jsonTemperatures` but, after a while (maybe 20-30-40 iterations) that nodes is starting to be empty

what is "strange" to me is that the `Serial.println(jsonOutput);` printing the serialzed of `jsonTemperatures` always returns proper data, the issue is limited to the `json_mqtt_state` json

maybe a full buffer somewhere ? how can I debug this ?

```

ewaldc, 2020-07-01:
Rapidly scanning your code, my guess would be that  _json_mqtt_state_ is running out of buffer space.
_jsonTemperatures_ is a local JsonDocument that is created each time again with proper sizing, so that explains why it is always OK.
However, at each iteration you are **copying** the whole _jsonTemperatures_ document to the DynamicJsonDocument _json_mqtt_state_, so it's running out of space as previous temperature data is not cleaned out.

A different approach will be needed, e.g. create _json_mqtt_state_ for each update rather than reusing it.

bblanchon, 2020-07-01:
See also:

* [I found a memory leak in the library!](https://arduinojson.org/v6/faq/i-found-a-memory-leak-in-the-library/)
* [Why must I create a separate config object? Why can't I use JsonDocument directly?](https://arduinojson.org/v6/faq/why-must-i-create-a-separate-config-object/)

guestisp, 2020-07-01:
Yes, solved by creating both JSON at the top and then clearing it every time

closed 2020-07-01 by bblanchon