bblanchon/ArduinoJson #1308: ArduinoJson Assistant 6: Calculates Capacity Incorrectly

lbussy, 2020-07-08:
When I use ArduinoJson Assistant 6 with the below JSON, my JSON truncates.  Empirically, by instantiating the DynamicJsonDocument with a larger mem size, the truncation does not happen.  Therefore I am assuming (I hope correctly) that there's something in the way I've formatted my JSON which the Assistant doesn't like:

Model used for Assistant:

``` json
{
  "imperial": false,
  "controlpoint": 99,
  "setting": 999.999,
  "status": 99,
  "controlenabled": false,
  "displayenabled": false,
  "sensors": [
    {
      "name": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
      "enable": false,
      "value": 999.999
    },
    {
      "name": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
      "enable": false,
      "value": 999.999
    },
    {
      "name": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
      "enable": false,
      "value": 999.999
    },
    {
      "name": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
      "enable": false,
      "value": 999.999
    },
    {
      "name": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
      "enable": false,
      "value": 999.999
    }
  ]
}
``` 

Expected:

``` json
{
  "imperial": true,
  "controlpoint": 4,
  "setting": 35,
  "status": 0,
  "controldisabled": false,
  "sensors": [
    {
      "name": "Room",
      "enable": false,
      "value": 85.1
    },
    {
      "name": "Tower",
      "enable": false,
      "value": 86
    },
    {
      "name": "Upper Chamber",
      "enable": false,
      "value": 78.8
    },
    {
      "name": "Lower Chamber",
      "enable": false,
      "value": 75.22813
    },
    {
      "name": "Keg",
      "enable": false,
      "value": 81.05
    }
  ],
  "displayenabled": false
}
```

Actual:

``` json
{
  "imperial": true,
  "controlpoint": 4,
  "setting": 35,
  "status": 0,
  "controldisabled": false,
  "sensors": [
    {
      "name": "Room",
      "enable": false,
      "value": 85.1
    },
    {
      "name": "Tower",
      "enable": false,
      "value": 85.1
    },
    {
      "name": "Upper Chamber",
      "enable": false,
      "value": 78.8
    },
    {
      "name": "Lower Chamber",
      "enable": false,
      "value": 75.2
    },
    {
      "name": "Keg"
    }
  ]
}
```

Code generating JSON:

``` cpp
    server.on("/sensors/", HTTP_ANY, [](AsyncWebServerRequest *request) {
        Log.verbose(F("Serving /sensors/." CR));
        const size_t capacityTempsSerial = JSON_ARRAY_SIZE(5) + 5*JSON_OBJECT_SIZE(3) + JSON_OBJECT_SIZE(7);
        DynamicJsonDocument doc(capacityTempsSerial );

        doc["imperial"] = config.copconfig.imperial;
        doc["controlpoint"] = config.temps.controlpoint;
        doc["setting"] = config.temps.setpoint;
        doc["status"] = tstat.state;

        if (config.temps.enabled[config.temps.controlpoint])
        {
            doc["controlenabled"] = true;
        }
        else
        {
            doc["controldisabled"] = false;
        }

        int numEnabled = 0;
        for (int i = 0; i < NUMSENSOR; i++)
        {
            doc["sensors"][i]["name"] = device.sensor[i].name;
            doc["sensors"][i]["enable"] = config.temps.enabled[i];
            if (config.temps.enabled[i])
            {
                numEnabled++;
            }
            if (config.copconfig.imperial)
            {
                doc["sensors"][i]["value"] = convertCtoF(device.sensor[i].average);
            }
            else
            {
                doc["sensors"][i]["value"] = device.sensor[i].average;
            }
        }

        if (numEnabled)
        {
            doc["displayenabled"] = true;
        }
        else
        {
            doc["displayenabled"] = false;
        }

        String json;
        serializeJsonPretty(doc, json);
        request->header("Cache-Control: no-store");
        request->send(200, F("application/json"), json);
    });
}
```

bblanchon, 2020-07-08:
Hi @lbussy,

The `JsonDocument` is probably saving a few strings in the memory pool.
Remember that only `const char*` are saved by pointer.

Best regards,
Benoit

lbussy, 2020-07-08:
Thanks for the reply Benoit.

Making sure I understand (and I probably missed this somewhere), you mean that:

``` cpp
        doc["imperial"] = config.copconfig.imperial;
        doc["controlpoint"] = config.temps.controlpoint;
        doc["setting"] = config.temps.setpoint;
        doc["status"] = tstat.state;
```

Should be:

``` cpp
        const bool imperial = config.copconfig.imperial;
        const int controlpoint = config.copconfig.imperial;
        const float setpoint = config.temps.setpoint;
        const int state = tstat.state;
        doc["imperial"] = imperial;
        doc["controlpoint"] = controlpoint;
        doc["setting"] = setpoint;
        doc["status"] = state;
```

Am I following that correctly?

ewaldc, 2020-07-08:
It's not the boolean's, integers, chars or floats that consume the extra memory, it the Strings and char pointers/arrays.
So indeed:
`doc["sensors"][i]["value"] = convertCtoF(device.sensor[i].average);`
or 
`            doc["sensors"][i]["name"] = device.sensor[i].name;`
are lines that can make the buffer pool run out of memory.
Keys will be copied in the buffer unless they are `const char *` or they can be modified in the source string (`char *`). 
"String" values will be copied to the buffer unless they are `const char *` as ArduinoJson assumes that if you use a String or '`char *`  these might be modified later in code and then the JsonDocument would no longer reflect the correct data.

lbussy, 2020-07-08:
Well, those are of course the more challenging ones to address.  I need to stare at it a while.  Thanks for the clarification Ewald.

lbussy, 2020-07-09:
I suspect I am in the right place with the wrong solution:

``` cpp
    server.on("/sensors/", HTTP_ANY, [](AsyncWebServerRequest *request) {
        Log.verbose(F("Serving /sensors/." CR));
        const size_t capacity = JSON_ARRAY_SIZE(5) + 5*JSON_OBJECT_SIZE(3) + JSON_OBJECT_SIZE(7);
        DynamicJsonDocument doc(capacity);

        doc["imperial"] = config.copconfig.imperial;
        doc["controlpoint"] = config.temps.controlpoint;
        doc["setting"] = config.temps.setpoint;
        doc["status"] = tstat.state;
        doc["controlenabled"] = config.temps.enabled[config.temps.controlpoint];

        int numEnabled = 0;
        char * sensorName[NUMSENSOR];
        double sensorAverage[NUMSENSOR];

        for (int i = 0; i < NUMSENSOR; i++)
        {
            sensorName[i] = device.sensor[i].name;
            doc["sensors"][i]["enable"] = config.temps.enabled[i];

            if (config.temps.enabled[i])
            {
                numEnabled++;
            }
            if (config.copconfig.imperial)
            {
                sensorAverage[i] = convertCtoF(device.sensor[i].average);
            }
            else
            {
                sensorAverage[i] = device.sensor[i].average;
            }
        }
        for (int i = 0; i < NUMSENSOR; i++)
        {
            doc["sensors"][i]["name"] = sensorName[i];
            doc["sensors"][i]["value"] = sensorAverage[i];
        }

        const bool displayenabled = (numEnabled > 0);
        doc["displayenabled"] = displayenabled;

        String json;
        serializeJsonPretty(doc, json);
        request->header("Cache-Control: no-store");
        request->send(200, F("application/json"), json);
    });
```

I think I addressed the average and the name properly, however I get the same results so I kinda expect I made a mistake. :)

bblanchon, 2020-07-09:
This program still inserts `char*`. To avoid the duplication, it must be a `const char*`.

You simply need to cast the pointer:

```c++
char value[] = "hello";
doc["value"] = (const char*)value; // <- no duplication
```

If you have a `String`, you can call `String::c_str()`:

```c++
String value = "hello";
doc["value"] = myString.c_str();  // <- no duplication
```

I don't usually recommend doing that because the `JsonDocument` could store a pointer to a destructed variable, but, in your case, it's safe because you immediately discard the `JsonDocument.`

lbussy, 2020-07-09:
I don't know why I never rememver to cast ... it's so easy.  Maybe that's why.  It's so easy it feels like cheating.

I'm good now, thank you both!

bblanchon, 2020-07-09:
You're welcome, Lee :-)

closed 2020-07-09 by bblanchon