bblanchon/ArduinoJson #1309: Enhancing the new Filtering with a wildcard

joysfera, 2020-07-11:
The cool filtering while deserializing introduced in v6.15 is great but it could be even better if there was a possibility to specify a wildcard for parents of nested documents if you need to filter somewhere deeper in the tree.

Imagine JSON document like this:
```
{
    "somekey": { "name": "some uninteresting name", "value": 2.718281828 },
    "otherkey": { "name": "guess what the value is", "value": 3.1415926535 },
    ...
    "xyzkey":  { "name": "ran out of interesting constants", "value": 1000 }
}
```

And I'd need to filter out the names (because they're long and consume a lot of precious RAM) and preserve just the key names (that unfortunately hold some important information) and the values inside of their objects (because the values are what matters).

So the Filtering document would need to look kind of like this:

`{ "*": { "value": true } }`

Would this be possible to implement easily and cleanly, please? Or any better idea?

Thank you!

closed 2020-09-30 by bblanchon

jshep321, 2021-01-30:
Great feature @bblanchon!  My wildcard filtering is not working exactly as expected and I'd really appreciate some feedback.
My input:
```
String fullUserDB = 
{
    "USER1": { "data" : { "1" : "item 1's description", 
                          "2" : "item 2's description"}, 
              "associated user data" : "value"},

    "USER2:  { "data" : { "10" : "item 10's description,
                          "11" : "item 11's description"} ,
              "associated user data" : "value"}
};
```
Say that I want to filter on the key "10".   I did this:
```
filter["*"]["data"]["10"] = true;
DeserializationError err = deserializeJson(jsonDoc, fullUserDB, DeserializationOption::Filter(filter));  //deserialize in order to apply filter
```
and I expected to retain only USER2's details and higher levels (including associated user data).  
What I was expecting:
``` 
{
    "USER2:  { "data" : { "10" : "item 10's description} ,
              "associated user data" : "value"}
}
```
However, I'm not keeping the "associated" data, but I am keeping some of USER1 details as below: 
Example result after above filtering (and re-serialization to a String and print):
{"USER1":{"data":{}},"USER2":{"data":{"10":"item 10's description}}}

Am I missing something or is the wildcard not as flexible as I was hoping?
Thanks!

bblanchon, 2021-01-31:
Hi @joysfera,

This is the expected behavior: the filter doesn't trim empty objects.
We could argue that this is a limitation of the library; however, your JSON document is very peculiar and not representative of most use cases.
If you need further assistance, please [open a new issue](https://github.com/bblanchon/ArduinoJson/issues/new/choose).

Best regards,
Benoit

joysfera, 2021-01-31:
FYI, it was @jshep321 :-)

jshep321, 2021-02-01:
Will do.  I will write a test program instead of manually edited json in the comments (as I have omitted several quotes above!).

commit 8385d5fa (2020-09-30): Added wildcard key (`*`) for filters (closes #1309)