bblanchon/ArduinoJson #1310: Data in JsonDocument getting stored as Boolean

mntolia, 2020-07-12:
Hi @bblanchon ,

I am not sure what I am doing wrong in the code below. Every key in the JSON document has the value of true when it is supposed to have a const char *. I ran each of the functions individually and it retuned the correct string. 
```cpp
 Serial.println("Creating Json Arrays");
    DynamicJsonDocument array(1024 * 50); //50 kb
    StaticJsonDocument<JSON_OBJECT_SIZE(4)> doc;
    JsonObject obj = doc.to<JsonObject>();
    //JsonArray array = arraydoc.to<JsonArray>();

    Serial.println("Inputting Reccuring keys");
    for (String i : recurring_keystore)
    {
      obj["time"] = fmem.getString((i + time_str).c_str());
      obj["days"] = get_days_in_str(i);
      obj["status"] = command_map(get_command_from_fmem((i).c_str()));
      obj["action"] = "/remove-time-recurring?unique_key=" + i;
      serializeJson(doc, Serial);
      array.add(obj);
    }
```
This prints to serial:
```
{{"time":true,"days":true,"status":true,"action":true}
```
I am using ArduinoJson 6.14.1.

Thanks for the library!

bblanchon, 2020-07-13:
Hi,

This weird behavior happens when [ARDUINOJSON_ENABLE_ARDUINO_STRING](https://arduinojson.org/v6/api/config/enable_arduino_string/) is 0.
Because the overload for `String` is missing, the compiler tries to select another one.
In this case, it decides to use the implicit conversion from `String` to `bool`.

In a similar situation, ArduinoJson 6.15+ triggers the following compiler error:

```c++
error: no matching function for call to 'VariantRef::set(const String&)'
```

I guess that's why you kept version 6.14 😉 

Best regards,
Benoit

mntolia, 2020-07-14:
It fixed the issue. I even updated it to 6.15.2. Thanks for the reply :)

Once I am done with my project I will purchase your book!

mntolia, 2020-07-14:
Hi @bblanchon , 
I am facing another issue in the above program. When I add more than 2 values to the JSONarray the remaining values get stored as null.

The output array comes like this:
```json
[{"time":"00:21","days":"Wed,Thu,","status":"ON","action":"/remove-time-recurring?unique_key=LZJn9"},{"time":"00:22","days":"Wed,Thu,","status":"OFF","action":null},{"time":null,"days":null,"status":"ON","action":null},{"time":null,"days":null,"status":null,"action":null},{"time":null,"days":null,"status":null,"action":null},{"time":null,"days":null,"status":null,"action":null},{"time":null,"days":null,"status":null,"action":null}]
```
But when I increase the StaticJsonDocument size to 50:
    StaticJsonDocument<JSON_OBJECT_SIZE(50) + 80> doc;

Then the output comes correctly. I read from your documentation that:
`If you pass a JsonArray, a JsonObject, or a JsonVariant, JsonVariant::set() makes a complete clone of the argument. In other words, the value is stored by copy, not by reference.`

So I assumed it is being copied. Then the values should not come as NULL. What could be wrong?

bblanchon, 2020-07-15:
Hi @mntolia,

Please see:

* [Why is the output incomplete?](https://arduinojson.org/v6/faq/why-is-the-output-incomplete/)

Best regards,
Benoit

closed 2020-07-20 by mntolia