bblanchon/ArduinoJson #1311: 'getOrAddElement': is not a member of ElementProxy::JsonDocument &>'

ewaldc, 2020-07-13:
* short description of the problem
This error is generated when compiling `src/extras/tests/ElementProxy/subscript.cpp `
* target platform
Windows 10 x64, AJ 6.15.2
* compiler model and version
Microsoft Visual Studio Enterprise 2017 Version 15.9.24 +28307.1177
Microsoft Visual C++ 2017
* MVCE (https://stackoverflow.com/help/mcve)
```
#include <ArduinoJson.h>
#include <catch.hpp>

using namespace ARDUINOJSON_NAMESPACE;

TEST_CASE("MemberProxy::operator[]") {
  DynamicJsonDocument doc(4096);
  ElementProxy<JsonDocument&> ep = doc[1];

  SECTION("set member") {
    ep["world"] = 42;

    REQUIRE(doc.as<std::string>() == "[null,{\"world\":42}]");
  }

  SECTION("set element") {
    ep[2] = 42;   // ---> **error is here**

    REQUIRE(doc.as<std::string>() == "[null,[null,null,42]]");
  }
}

```
* compiler output
Error C2039 'getOrAddElement': is not a member of 'ArduinoJson6152_1100000::ElementProxy<ArduinoJson6152_1100000::JsonDocument &>'	ElementProxyTest	c:\users\ewald\documents\visual studio 2017\projects\openhabesp_test\src\arduinojson\array\elementproxy.hpp	163	

As I was validating my 5.x port for relocatable VariantSlot onto v6.x, I stumbled on this test that refused to compile.  When substituted for the original AJ 6.15.2, it seems the compiler generates the same error.
Can't really understand what is wrong, since `JsonDocument.hpp` contains:
```
  FORCE_INLINE VariantRef getOrAddElement(size_t index) {
    return VariantRef(&_pool, _data.getOrAddElement(index, &_pool));
  }
```
but I have to admit I don't full understand the whole purpose of this code.  Normally I would just add a JsonArray to the  JsonDocument, but I assume this is specifically testing ElementProxy::JsonDocument &

ewaldc, 2020-07-15:
Closing for now.  6.15.1 has other challenges that are solved in 6.15.2 (comparisons).  I will just skip this test...

bblanchon, 2020-07-23:
Thanks a lot, @ewaldc.

closed 2020-07-23 by bblanchon

bblanchon, 2020-08-01:
This fix is available in [ArduinoJson 6.16.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.16.0).

commit 6dc36125 (2020-07-23): Added ElementProxy::getOrAddElement() (fixes #1311)