bblanchon/ArduinoJson #1313: Optimizing memory allocation when de-serializing input of unknown max. size

dyarkovoy, 2020-07-17:
I know this has been discussed many times, and is covered in the docs and related issues.
Still, parsing SPIFFS-based config is somewhat challenging for the cases when estimating the maximum possible size of the document is non-trivial (for example, when the application is modular and the config is centralized, and each module can have its own entry in the global config). I suspect the recommedations will be to have each module either maintain its own config, or make the module compute max.memory usage.

But I wanted to see if we can go a different route. 

Given a JsonDocument, can we compute the **exact** amount of memory needed to restore it from the const char* or a stream? 
Then we can just write this number in the first bytes of the config file and create DynamicJsonDocument with optimal size when reading, problem solved!

At first I thought that JsonDocument::memoryUsage() would do the trick. but no. The initial config tree may contain references to strings not owned by the document buffer, and in these cases we would need a larger buffer to de-serialize. JsonDocument::capacity() is a little better, but still suffers from the same issue if the creator of the document deliberately decreased max. size knowing that some strings will be referenced from the "out there".

_So my question is: can we have a method that would walk the document tree and compute the memory usage **as if** all the strings were contained in the document buffer?_

ewaldc, 2020-07-17:
On ESP platform, I solve this as follows for files (and similarly for web posts):
```
DynamicJsonDocument OpenHab::getJsonDocFromFile(String fileName) {
	File f = SPIFFS.open(fileName, "r");
	DynamicJsonDocument doc(ESP.getMaxFreeBlockSize() - 4096);
	DeserializationError err = deserializeJson(doc, f, DeserializationOption::NestingLimit(20));
	if (err == DeserializationError::Ok) doc.shrinkToFit();
	else DbgPrintln(F("deserializeJson failed "), err.c_str());
	f.close();
	return doc;
}
```
There are alternative ways to estimate the initial size e.g. 2x the Json String, but this works well and does not result in too much memory fragmentation if the _JsonDocuments_ are temporary in nature.
Works flawlessly with literally thousands of unknown strings/day submitted over the web (and one reboot every week... but not because of AJ).  The owner of this library was so kind to add the _shrinkToFit()_ for this purpose...

dyarkovoy, 2020-07-17:
I like this approach and I see how it works well, but couldn't we replace the huge allocation and shrinkToFit() by computing the exact size of the document before it gets written?
If you are on ESP32 with multiple tasks running, I think you may run into occasional out of memory errors if some other task interfere and want to allocate more than 4096 bytes. Need to wrap in mutex or use some other synchronization to avoid that potential issue.

dyarkovoy, 2020-07-17:
So I went ahead and wrote a quick and dirty function to measure document memory usage regardless of the string references (based on the SizeAccumulatorV6 @ https://arduinojson.org/v6/assistant/). Works well for me so far, here it is just in case anyone wants it:
```
    size_t measure(const JsonVariantConst v)
    {
      size_t result = 0;
      if (v.is<JsonArray>())
      {
        JsonArrayConst a = v.as<JsonArrayConst>();
        result += JSON_ARRAY_SIZE(a.size());
        for (JsonVariantConst i : a)
          result += measure(i);
      }
      else if (v.is<JsonObject>())
      {
        JsonObjectConst o = v.as<JsonObjectConst>();
        result += JSON_OBJECT_SIZE(o.size());
        for (JsonPairConst kv : o)
          result += JSON_STRING_SIZE(strlen(kv.key().c_str())) + 1 + measure(kv.value());
      }
      else if (v.is<const char *>())
      {
        const char *c = v.as<const char *>();
        result += JSON_STRING_SIZE(strlen(c)) + 1;
      }
      return result;
    }
```

ewaldc, 2020-07-17:
Maybe I don't understand what you want to accomplish....

If you parse from a file, const char * or stream than **all** strings will copied in the JsonDocument. so _measureJson()_ will yield the correct memory consumption and there is no need for code that recursively scans the JsonDocument.

The more difficult issue IMHO is to decide how big to size your JsonDocument for _deserialize()_ when dealing with "unknown" Json source strings e.g. from config files  i.e. the code you posted will do the job, but it requires that you already have deserialized the stream/file/const char*...

dyarkovoy, 2020-07-17:
> The more difficult issue IMHO is to decide how big to size your JsonDocument for _deserialize()_ when dealing with "unknown" Json source strings e.g. from config files i.e. the code you posted will do the job, but it requires that you already have deserialized the stream/file/const char*...

This.
* I serialize document containing app config to SPIFFS.
* Config may vary in size greatly and I don't want to pre-allocate huge chunk of RAM to accommodate all possible (very long and very short) JSON strings when deserializing.
* At the time of **serializing** I know the exact size of the document and exact amount of RAM required to de-serialize.
* At the time of **serializing** I compute this exact amount of RAM and store it in the first 4 bytes of the config file.
* At the time of **de-serializing** I read this number from the first 4 bytes of the config file and pass it to DynamicJsonDocument() constructor and then proceed with de-serialization of the rest of the file.
This way I avoid allocation of big buffer and subsequentaly calling shrinkToFit().
My problem was to find the right algorithm to compute the exact size of the document, because sometimes it owns strings, and sometimes not (see explanation in my first post). Therefore I suggested a function that solves this problem and computes the size of the document regardless of the string being owned or not.
Hope this makes things clear.

bblanchon, 2020-07-17:
Hi @dyarkovoy,

I agree with @ewaldc, `getMaxFreeBlockSize()` and [`shrinkToFit()`](https://arduinojson.org/v6/api/basicjsondocument/shrinktofit/) is the best solution to your problem.

You're right, it will temporarily use more memory than strictly required, but it's just for a fraction of a second.
I really doubt it could interfere with a concurrent task, especially if you're using this code to load a configuration file at startup.

ArduinoJson 6 was designed to fight against heap fragmentation while keeping a minimal code size, and fixed allocation represents an excellent compromise. Elastic capacity will come back, but not in the foreseeable future.

Best regards,
Benoit

ewaldc, 2020-07-17:
Ah, I see.  If you are the one serializing every single JSON string to a file, then you can of course save the consumed memory with the file.  And if the Json documents are composed in bits and pieces (e.g. a mix of const char * and char *), than your code will calculate the exact size required for deserialization from file/const char *.
Most of us don't have the luxury to deal with all known, pre-scanned JSON documents...

dyarkovoy, 2020-07-18:
Hi, @bblanchon, I understand and agree, I was just offering an alternative that  doesn't require` getMaxFreeBlockSize()` and `shrinkToFit()`. Here is how I do it:

```
  void ConfigFS::write(const DynamicJsonDocument& config)
  {
    File file = _fs.open(_path, FILE_WRITE);
    if (file)
    {
      size_t mu = json::measure(config.as<JsonVariantConst>());
      file.write((uint8_t *)&mu, sizeof(mu));
      serializeJson(config, file);
      file.flush();
      file.close();
    }
  }

DynamicJsonDocument* ConfigFS::read()
  {
    DynamicJsonDocument *doc = nullptr;
    File file = _fs.open(_path);
    if (file)
    {
      size_t mu = 0;
      if (file.read((uint8_t *)&mu, sizeof(mu)) == sizeof(mu) && mu)
      {
        doc = new DynamicJsonDocument(mu);
        auto r = deserializeJson(*doc, file);
        if (r != DeserializationError::Ok)
        {
          delete doc;
          doc = nullptr;
        }
      }
      file.close();
    }
    return doc;
  }
```

In my view, this is better as it doesn't require huge allocation and then re-allocation with `shrinkToFit()`

closed 2020-07-29 by bblanchon