bblanchon/ArduinoJson #1314: Heap corruption by deserializeJson()

dyarkovoy, 2020-07-17:
Environment:
PlatformIO 4.4.05a (development version) on VSCode
ESP32 dev board
framework = arduino, espidf

Repro: [https://github.com/esp32m/bugs/tree/master/aj-bug](https://github.com/esp32m/bugs/tree/master/aj-bug)

Notes:
It may or may not be related to ArduinoJson, as it happens only under very specific conditions:
* v4.0 of the framework-arduinoespressif32
* arduino used as library in esp-idf (framework = arduino, espidf)
* the document is de-serialized from the Arduino Stream (SPIFFS)
* DynamicJsonDocument is created with a small buffer and deserializeJson returns NoMemory
* -DARDUINOJSON_USE_DOUBLE=0 - not strictly necessary, but corrupts the heap faster with this setting

This is what I get when running the code:
```

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7016
load:0x40078000,len:14076
load:0x40080400,len:4304
entry 0x400806e8
I (71) boot: Chip Revision: 1
I (72) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot: ESP-IDF HEAD-HASH-NOTFOUND 2nd stage bootloader
I (39) boot: compile time 09:44:08
I (40) boot: Enabling RNG early entropy source...
I (45) boot: SPI Speed      : 40MHz
I (49) boot: SPI Mode       : DIO
I (53) boot: SPI Flash Size : 4MB
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (76) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (83) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  3 ota_0            OTA app          00 10 00010000 001f0000
I (98) boot:  4 ota_1            OTA app          00 11 00200000 001f0000
I (106) boot:  5 eeprom           Unknown data     01 99 003f0000 00001000
I (113) boot:  6 spiffs           Unknown data     01 82 003f1000 0000f000
I (121) boot: End of partition table
I (125) boot_comm: chip revision: 1, min. application chip revision: 0
I (132) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x08314 ( 33556) map
I (153) esp_image: segment 1: paddr=0x0001833c vaddr=0x3ffb0000 size=0x0240c (  9228) load
I (157) esp_image: segment 2: paddr=0x0001a750 vaddr=0x40080000 size=0x00400 (  1024) load
I (162) esp_image: segment 3: paddr=0x0001ab58 vaddr=0x40080400 size=0x054b8 ( 21688) load
I (179) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x20794 (133012) map
I (227) esp_image: segment 5: paddr=0x000407b4 vaddr=0x400858b8 size=0x050d4 ( 20692) load
I (243) boot: Loaded app from partition at offset 0x10000
I (243) boot: Disabling RNG early entropy source...
I (244) cpu_start: Pro cpu up.
I (247) cpu_start: Application information:
I (252) cpu_start: Project name:     aj-bug
I (257) cpu_start: App version:      1
I (261) cpu_start: Compile time:     Jul 17 2020 09:43:44
I (267) cpu_start: ELF file SHA256:  361885746a3b8d98...
I (273) cpu_start: ESP-IDF:          HEAD-HASH-NOTFOUND
I (279) cpu_start: Starting app cpu, entry point is 0x400814ac
I (266) cpu_start: App cpu up.
I (290) heap_init: Initializing. RAM available for dynamic allocation:
I (297) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (303) heap_init: At 3FFB3538 len 0002CAC8 (178 KiB): DRAM
I (309) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (315) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (322) heap_init: At 4008A98C len 00015674 (85 KiB): IRAM
I (328) cpu_start: Pro cpu start user code
I (347) spi_flash: detected chip: generic
I (347) spi_flash: flash io: dio
I (348) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
trying size 200CORRUPT HEAP: Block 0x73227b38 is outside heap (last valid block 0x3ffb8af0)
heap corrupted aftr calling deserializeJson()
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4000c789  PS      : 0x00060030  A0      : 0x80001d53  A1      : 0x3ffb5a40
A2      : 0x00000000  A3      : 0x40001fec  A4      : 0x3ffb4324  A5      : 0x00000001
A6      : 0x3ffb43af  A7      : 0x3ffb424c  A8      : 0x00060023  A9      : 0x00000001
A10     : 0x00060021  A11     : 0x00060f23  A12     : 0x00060f20  A13     : 0x00000001
A14     : 0x00060f20  A15     : 0xfffffffc  SAR     : 0x00000000  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000000d  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

ELF file SHA256: 361885746a3b8d98

Backtrace: 0x4000c786:0x3ffb5a40 0x40001d50:0x3ffb5a60 0x400e0c69:0x3ffb5a80 0x400869b6:0x3ffb5aa0 0x4008763a:0x3ffb5ac0 0x4008764f:0x3ffb5ae0 0x40085ad9:0x3ffb5b00
  #0  0x4000c786:0x3ffb5a40 in ?? ??:0
  #1  0x40001d50:0x3ffb5a60 in ?? ??:0
  #2  0x400e0c69:0x3ffb5a80 in cbuf::write(char const*, unsigned int) at C:\Users\user\.platformio\packages\framework-arduinoespressif32-src-c69fc9322995db5694e2741d307d9723\cores\esp32/cbuf.h:68
      (inlined by) cbuf::write(char const*, unsigned int) at C:\Users\user\.platformio\packages\framework-arduinoespressif32-src-c69fc9322995db5694e2741d307d9723\cores\esp32/cbuf.cpp:171
  #3  0x400869b6:0x3ffb5aa0 in tx_pwctrl_cal at /home/aiqin/git_tree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_cal.c:2645
  #4  0x4008763a:0x3ffb5ac0 in bootloader_flash_dummy_config at C:\Users\user\.platformio\packages\framework-espidf\components\bootloader_support\src/bootloader_flash_config.c:163
  #5  0x4008764f:0x3ffb5ae0 in esp_reset_reason_set_hint at C:\Users\user\.platformio\packages\framework-espidf\components\esp32/reset_reason.c:106
  #6  0x40085ad9:0x3ffb5b00 in bt_track_pll_cap at /home/aiqin/git_tree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:1440

Rebooting...
```

bblanchon, 2020-07-17:
Hi @dyarkovoy,

Thank you very much for reporting this issue.

Do you think you can reproduce this bug without SPIFFS, for example with a [StringStream](https://github.com/bblanchon/ArduinoStreamUtils#reading-from-a-string)?

BTW, this bug slightly resembles #1210, can you check that `-Og` is not used?

Best regards,
Benoit

bblanchon, 2020-07-17:
> it happens only under very specific conditions:

Can you tell us which other conditions you tested?

dyarkovoy, 2020-07-18:
Hi @bblanchon, thank you for the response. 

> Do you think you can reproduce this bug without SPIFFS, for example with a StringStream?

I was testing with Arduino's StreamString (framework-arduinoespressif32/cores/esp32/StreamString.h) and wasn't able to reproduce. It doesn't necessarily mean that the problem is related to SPIFFS, though. I noticed that even slight changes in the code, addition of libraries etc., result in a different behavior of the bug.

> BTW, this bug slightly resembles #1210, can you check that -Og is not used?

Indeed, I was under the impression that these bugs may be of the same nature. I confirm that -Og is used by default by PlatformIO build process even if I do release build. I wasn't able to remove the -Og using the build_unflags = -Og, it causes a number of other compilation errors in esp-idf

> Can you tell us which other conditions you tested?

The other conditions is my rather large project that was exhibiting heap corruption occasionally, and I traced it down to deserializeJson() after it returns NoMemory. Unfortunately it cannot be reproduced reliably within the scope of the project as the corruption is unpredictable. But I isolated the issue and created a separate project that reliably reproduces the problem. You can follow these steps to reproduce it:
* Install the latest version of VSCode and PlatformIO 4.4.05a (development version) on Windows 10 x64, installation instructions are here: https://docs.platformio.org/en/latest/core/installation.html#development-version
* Download the project reproducing the bug here: https://github.com/esp32m/bugs/tree/master/aj-bug
* Compile and upload to ESP32 chip, I used the most common one with 4MB Flash, no PSRAM
You should get the picture shown in my first post in the console.

ewaldc, 2020-07-18:
I have seen (and reported) corruptions with SPIFFS write on both ESP32 and ESP8266.  Not all all heap corruptions though, some are (flash) implementation specific.  It improves much by formatting the SPIFFS before starting to write to it.
Things have gotten more complicated now that SPIFFS is deprecated (e.g. patches I submitted are no longer being looked at)

dyarkovoy, 2020-07-18:
Thanks, @ewaldc, I just learned that SPIFFS is deprecated. It seems to be actively discussed by the ESP8266 community, but almost no mention of it in ESP32 (where I do most of the coding). So is FATFS now the way to go?

ewaldc, 2020-07-18:
@dyarkovoy, the code you posted first writes the file, then reads from it a number of times.   It is the write that under certain conditions can corrupt the SPIFFS structure and result in corruptions.    It does not happen though when you upload the SPIFFS file/file system during flash.  That particular issue I have not seen on ESP32 though, only ESP8266...
On my production code which parses 1000's of Json strings/day with AJ , I had to abandon write to SPIFFS, unless I was able to do a clean format first.  I now do my own writing to EEPROM (separate area).

dyarkovoy, 2020-07-18:
I see. I just commented out the part that writes to SPIFFS. Still get the same problem when reading.
Then, I added the SPIFFS.format(); at the start, wrote to SPIFFS just once, and still get the problem when reading.
Finally, I wiped out flash,, then uploaded a sketch that formatted SPIFFS and wrote JSON string, then uploaded separate sketch that did only the reading (no writing at all), and I still get the heap corruption when deserializeJson() is called.

ewaldc, 2020-07-18:
OK, that eliminates SPIFFS IMHO.  A area where I have seen AJ corrupt the heap with deserialization is _allocExpandableString()_
That string was/is not checked for bounds and it possibly runs into the VariantSlot pool.  When that happens pointers are being corrupted, and if you work with the JsonDocument afterwords it generally crashes, but it might corrupt memory too when maintaining the linked list of object elements. Your sample JSON string with a single nested root object is susceptible to this.
For that reason, in my production code, I used to reboot the ESP on failed deserialization due to shortage of memory.

I modified AJ in that respect (moved to a relocatable memory pool without pointers) which I hope to contribute back once ported to 6.x.
AJ is extremely well written/performing and robust, but recovering from every out of memory situation is a very challenging task...

bblanchon, 2020-07-18:
If it not related to SPIFFS, then there must be a way to reproduce it without SPIFFS.

I really doubt that the corruption comes from `deserializeJson()` because I'm running [continuous fuzzing with address sanitizer, memory sanitizer, and undefined behavior sanitizer](https://github.com/google/oss-fuzz/blob/master/projects/arduinojson/project.yaml).

@dyarkovoy, you said that `-Og` is used even in release mode.
That's very strange.
Are you sure about that?
If it's true, then we're probably hitting #1210 again, and we must find a way to remove this flag.

dyarkovoy, 2020-07-20:
Ok. My problem is solved for now by ensuring that `deserializeJson()` never runs out of memory.

>@dyarkovoy, you said that -Og is used even in release mode.
That's very strange.
Are you sure about that?

Please take a look:
```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.12.4 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 2bf655b (git+https://github.com/espressif/arduino-esp32.git#idf-release/v4.0)
 - framework-espidf 3.40001.200521 (4.0.1)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.190618 (2.28.51)
 - toolchain-xtensa32 2.80200.200226 (8.2.0)
Reading CMake configuration...
Platform incompatible library C:\Users\user\.platformio\lib\ESPAsyncTCP
More details about "Library Compatibility Mode": https://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Platform incompatible library C:\Users\user\.platformio\lib\ESPAsyncTCP
Found 18 compatible libraries
More details about "Library Compatibility Mode": https://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson> 6.15.2 (C:\Users\user\.platformio\lib\ArduinoJson)
Building in release mode
xtensa-esp32-elf-g++ -o .pio\build\esp32dev\esp-idf\src\main.cpp.o -c -std=gnu++11 -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fno-exceptions -fno-rtti -DPLATFORMIO=40400 -DARDUINO_ESP32_DEV -DARDUINOJSON_USE_DOUBLE=0 -D_GNU_SOURCE -DIDF_VER=\"HEAD-HASH-NOTFOUND\" -DGCC_NOT_5_2_0 -DESP_PLATFORM @"D:\arduino\projects\bugs\aj-bug\.pio\build\esp32dev\longcmd-459f156384cb3c2d7f369f72743c7b3e" src\main.cpp
<lambda>(["checkprogsize"], [".pio\build\esp32dev\firmware.elf"])
MethodWrapper(["checkprogsize"], [".pio\build\esp32dev\firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   4.2% (used 13612 bytes from 327680 bytes)
Flash: [=         ]  10.8% (used 219197 bytes from 2031616 bytes)

```
> and we must find a way to remove this flag.

When I remove -Og, it hits a number of errors during esp-idf compilation. I will see if I can fix these when I have time, or maybe the esp-idf developers fix them by the time.

dyarkovoy, 2020-07-20:
Actually I was able to recompile by replacing -Og with -O1 and  tweaking some build flags. The heap corruption is gone now.
Also tried with -O2, -O3, -Os - works smoothly. Couldn't find a way to build without -O or with -O0  - the compiler bloats the size of the bootloader.bin and it doesn't fit in the somewhere-hardcoded-limit (increasing CONFIG_PARTITION_TABLE_OFFSET doesn't help unfortunately).

So now that we are almost positive about the cause, should we report this issue to the esp-idf devs?

ewaldc, 2020-07-20:
I have tried to report it without success. Trouble was that I have not been able to reproduce it reliably without AJ and/or Arduino IDE (only PlatformIO).  They asked to test with different versions of the compiler, but that brings in different versions of Arduino ESP and new problems (e.g. linking problems, map problems etc.).   And strange enough all crashes I recorded were after a failed de-serialization, even though I think the only role AJ plays is to trigger the defect.
Also strange things are happening once you install the Platformio development version e.g. not able to go back to standard version: there is always some difference as I compared the binaries produced by two systems (on standard install versus rolled back install).   But then again, I use PlatformIO in portable version...

bblanchon, 2020-07-20:
I wasn't able to reproduce the problem without ArduinoJson, but I didn't push very far.
I think the ingredients we need are: 1. recursion, 2. templates
I'll try to find a moment to retry, but I'm very busy these days.

bblanchon, 2020-07-26:
I just closed #1210.
The fix is available on branch `6.x`.

closed 2020-07-26 by bblanchon

dyarkovoy, 2020-07-26:
**Confirmed!** 
6.15.2 #144ff3b doesn't cause heap corruption with -Og
Thank you, @bblanchon !

bblanchon, 2020-08-01:
This fix is available in [ArduinoJson 6.16.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.16.0).

commit 144ff3b0 (2020-07-26): Updated the change log. Closes #1210 and closes #1314.