bblanchon/ArduinoJson #1316: Filtering nested JSON

ArnieO, 2020-07-18:
**_Bonjour Benoît!_**
I'm working on the below JSON (API link: [https://api.apispreadsheets.com/data/782](https://api.apispreadsheets.com/data/782))

I'd like to filter out one of the five sections, representing data for one region.
Using http://www.jsonquerytool.com/ I am able to do that with a JavaScript query `input["data"][2]`
But I have not succeeded in finding how to obtain the same using the filtering function.
I've tested various alternatives using [https://wandbox.org/permlink/KKc8gjsesx6yru7V](https://wandbox.org/permlink/KKc8gjsesx6yru7V) and what I find is:

- `jfilter["data"][0] = true;` returns the entire JSON
- `jfilter["data"][1] = true;` returns this :`
{
  "data": []
}
`

I'd greatly appreciate your guidance!

_JSON example data, in case above API link no longer works:_
```{
  "data": [
    {
      "1": "0,01547527",
      "2": "0,015369275",
      "3": "0,01526328",
      "4": "0,01526328",
      "5": "0,015157285",
      "6": "0,015157285",
      "7": "0,015157285",
      "8": "0,015157285",
      "9": "0,015369275",
      "10": "0,015581265",
      "11": "0,015793255",
      "12": "0,015793255",
      "13": "0,015793255",
      "14": "0,01568726",
      "15": "0,01568726",
      "16": "0,01568726",
      "17": "0,01568726",
      "18": "0,01568726",
      "19": "0,015793255",
      "20": "0,015793255",
      "21": "0,01568726",
      "22": "0,01568726",
      "23": "0,01568726",
      "24": "0,015581265",
      "Region": "NO-1",
      "Oppdatert": "20-07-18T11:27:18Z",
      "Prisdato": "2020-07-18",
      "Kilde": "https://transparency.entsoe.eu",
      "Enhet": "NOK/kWh"
    },
    {
      "1": "0,01547527",
      "2": "0,015369275",
      "3": "0,01526328",
      "4": "0,01526328",
      "5": "0,015157285",
      "6": "0,015157285",
      "7": "0,015157285",
      "8": "0,015157285",
      "9": "0,015369275",
      "10": "0,015581265",
      "11": "0,015793255",
      "12": "0,015793255",
      "13": "0,015793255",
      "14": "0,01568726",
      "15": "0,01568726",
      "16": "0,01568726",
      "17": "0,01568726",
      "18": "0,01568726",
      "19": "0,015793255",
      "20": "0,015793255",
      "21": "0,01568726",
      "22": "0,01568726",
      "23": "0,01568726",
      "24": "0,015581265",
      "Region": "NO-2",
      "Oppdatert": "20-07-18T11:27:19Z",
      "Prisdato": "2020-07-18",
      "Kilde": "https://transparency.entsoe.eu",
      "Enhet": "NOK/kWh"
    },
    {
      "1": "0,03519034",
      "2": "0,034448375",
      "3": "0,0339184",
      "4": "0,03370641",
      "5": "0,033600415",
      "6": "0,03349442",
      "7": "0,03370641",
      "8": "0,03370641",
      "9": "0,03476636",
      "10": "0,0360383",
      "11": "0,03752223",
      "12": "0,037416235",
      "13": "0,03667427",
      "14": "0,035932305",
      "15": "0,035932305",
      "16": "0,03561432",
      "17": "0,035932305",
      "18": "0,03646228",
      "19": "0,035720315",
      "20": "0,035084345",
      "21": "0,045471855",
      "22": "0,045047875",
      "23": "0,044623895",
      "24": "0,04472989",
      "Region": "NO-3",
      "Oppdatert": "20-07-18T11:27:20Z",
      "Prisdato": "2020-07-18",
      "Kilde": "https://transparency.entsoe.eu",
      "Enhet": "NOK/kWh"
    },
    {
      "1": "0,03519034",
      "2": "0,034448375",
      "3": "0,0339184",
      "4": "0,03370641",
      "5": "0,033600415",
      "6": "0,03349442",
      "7": "0,03370641",
      "8": "0,03370641",
      "9": "0,03476636",
      "10": "0,0360383",
      "11": "0,03752223",
      "12": "0,037416235",
      "13": "0,03667427",
      "14": "0,035932305",
      "15": "0,035932305",
      "16": "0,03561432",
      "17": "0,035932305",
      "18": "0,03646228",
      "19": "0,035720315",
      "20": "0,035084345",
      "21": "0,045471855",
      "22": "0,045047875",
      "23": "0,044623895",
      "24": "0,04472989",
      "Region": "NO-4",
      "Oppdatert": "20-07-18T11:27:21Z",
      "Prisdato": "2020-07-18",
      "Kilde": "https://transparency.entsoe.eu",
      "Enhet": "NOK/kWh"
    },
    {
      "1": "0,01547527",
      "2": "0,015369275",
      "3": "0,01526328",
      "4": "0,01526328",
      "5": "0,015157285",
      "6": "0,015157285",
      "7": "0,015157285",
      "8": "0,015157285",
      "9": "0,015369275",
      "10": "0,015581265",
      "11": "0,015793255",
      "12": "0,015793255",
      "13": "0,015793255",
      "14": "0,01568726",
      "15": "0,01568726",
      "16": "0,01568726",
      "17": "0,01568726",
      "18": "0,01568726",
      "19": "0,015793255",
      "20": "0,015793255",
      "21": "0,01568726",
      "22": "0,01568726",
      "23": "0,01568726",
      "24": "0,015581265",
      "Region": "NO-5",
      "Oppdatert": "20-07-18T11:27:22Z",
      "Prisdato": "2020-07-18",
      "Kilde": "https://transparency.entsoe.eu",
      "Enhet": "NOK/kWh"
    }
  ]
}
```

bblanchon, 2020-07-18:
Hi @ArnieO,

Unfortunately, the filtering feature doesn't allow selecting an element in an array.
Currently, it only allows filtering members in objects.

As a workaround, I suggest that you use the ["deserialization in chunks" technique](https://arduinojson.org/v6/how-to/deserialize-a-very-large-document/#deserialization-in-chunks).

I wish ArduinoJson could support array filtering, but I could not find an API that supports all use cases.
So far, people requested to:

- keep all of the elements of the array (current behavior)
- keep one element (as you asked)
- keep the n first elements
- keep the n last elements
- keep several elements at random locations.

Suggestions are welcome.

Best regards,
Benoit

ArnieO, 2020-07-18:
Bonjour @bblanchon,

Thank you for your rapid answer. I realize there must be many and varying wishes for filtering methods, and that it could increase size of embedded code if implemented. So it is a understandable approach to keep this functionality on a basic level. 

I looked at the "deserialization in chunks" method.
In my case, I have control of the API and instead modified it so I can select which element to read. So this was the best solution in this case. 

Best regards,

closed 2020-07-18 by ArnieO

bblanchon, 2020-07-19:
Thanks for the update.
Good luck with your project, @ArnieO.