bblanchon/ArduinoJson #1317: DeserializeJson with Wire. follow on from issue #1099

mishave, 2020-07-20:
Hey, having issues with trying to use the library with i2c. using serial it works a treat but over i2c I can't seem to get it to work.

## My Master Code is:
```#include <Wire.h>
#include <ArduinoJson.h>
#include <StreamUtils.h>

const size_t capacity = JSON_OBJECT_SIZE(9) + 176; //+176 use tool https://arduinojson.org/v6/assistant/
DynamicJsonDocument doc(capacity);

void setup()
{
  Wire.begin();        // join i2c bus (address optional for master)
  Serial.begin(9600);  // start serial for output
  for (int i = 2; i <= 8; i++) {
    pinMode(i, OUTPUT);
    digitalWrite(i, LOW);
    delay(10);
  }
  for (int i = 26; i <= 34; i++) {
    pinMode(i, OUTPUT);
    digitalWrite(i, LOW);
    delay(10);
  }
  pinMode(12, OUTPUT);
  digitalWrite(12, LOW);

}

void loop()
{
  Wire.requestFrom(2, capacity);    // request 6 bytes from slave device #2
    DynamicJsonDocument doc(capacity);
    // const char* json = "{\"Sensor Board 01\":\"Inputs\",\"Input01\":1234,\"Input02\":1234,\"Input03\":1234,\"Input04\":1234,\"Input05\":1234,\"Output01\":100000,\"Output02\":1234}";
    DeserializationError err = deserializeJson(doc, Wire);
    if (err) {
      Serial.print(F("deserializeJson() failed with code "));
      Serial.println(err.c_str());
    }
  const char* Sensor_Board_01 = doc["Sensor Board 01"]; // "Inputs"
  int Input01 = doc["Input01"]; // 1234
  int Input02 = doc["Input02"]; // 1234
  int Input03 = doc["Input03"]; // 1234
  int Input04 = doc["Input04"]; // 1234
  int Input05 = doc["Input05"]; // 1234
  int Output01 = doc["Output01"]; // 1000
  int Output02 = doc["Output02"]; // 1000

}
```

## While my Slave is:
```
#include <Wire.h>
#include <ArduinoJson.h>
const size_t capacity = JSON_OBJECT_SIZE(8);
DynamicJsonDocument doc(capacity);

float sensorValue[5];

int ledPin = 13;      // select the pin for the LED

void setup()
{
  //Start Serial for debugging
  Serial.begin(9600);
  //Set Up i2c Comms
  Wire.begin(2);                // join i2c bus with address #2
  Wire.onRequest(requestEvent); // register event
}

void loop() {
  readInputs();
  JSON();
  delay(100);
}

void readInputs() {
  for (int i = 0; i <= 5; i++) {
    sensorValue[i] = analogRead(i);
  }
}

void JSON() {
  doc["Sensor Board 01"] = "IO Status";
  doc["Input01"] = 1273;
  doc["Input02"] = 1270;
  doc["Input03"] = 1280;
  doc["Input04"] = 1277;
  doc["Input05"] = 1019;
  doc["Output01"] = 1234;
  doc["Output02"] = 1234;
}

// function that executes whenever data is requested by master
// this function is registered as an event, see setup()
void requestEvent() {
  serializeJson(doc, Wire);
  serializeJsonPretty(doc, Serial); //Print for debugging

}

```
The error code it is spitting out is: deserializeJson() failed with code IncompleteInput

Which is because the slave does not seem to register the request. not too sure where to go from here.

mishave, 2020-07-20:
Just for the hell of it I also tried the same code as the #1099 to see what the issue was on their end, it seems to be the same problem, calling
```` 
Wire.requestFrom(8, capacity);
  DynamicJsonDocument doc(capacity);
 ```
doesn't register the request on slaves end. could this be an issue of size? and instead (max 32Bytes), I'd need to break the message down into blocks?

mishave, 2020-07-20:
just playing around with it some more I've been using the code from #1099 as its shorter so quicker to play with,

For the master I have the following, please note I included while(Wire.available()) to scan the port and print what's there.
```
#include <Wire.h>
#include <ArduinoJson.h>

const size_t capacity = JSON_OBJECT_SIZE(3)+72;
void setup(){
  Wire.begin(8); 
  Serial.begin(9600);
  Serial.println("setup::done");
}

void loop() {
  delay(50);
  Wire.beginTransmission(8); 
  Wire.write("Hello Arduino");
  Wire.endTransmission(); 

  Wire.requestFrom(8, capacity);
  while(Wire.available())    // slave may send less than requested
  { 
    char c = Wire.read(); // receive a byte as character
    Serial.print(c);         // print the character
  }
  Serial.println(" ");

  DynamicJsonDocument doc(capacity);
  DeserializationError err = deserializeJson(doc, Wire);
  switch (err.code()) {
    case DeserializationError::Ok:
        Serial.println(F("Deserialization succeeded"));\
        serializeJson(doc, Serial);
        break;
    case DeserializationError::InvalidInput:
        Serial.println(F("Invalid input!"));
        break;
    case DeserializationError::NoMemory:
        Serial.println(F("Not enough memory"));
        break;
    default:
        Serial.println(F("Deserialization failed"));
        break;
  }
 
}
```
Slave I have
```
#include <Wire.h>
#include <ArduinoJson.h>

const size_t capacity = JSON_OBJECT_SIZE(3);

void setup(){
  Wire.begin(8);
  Wire.onRequest(requestEvent); /* register receive event */
  Wire.onRequest(requestEvent); /* register request event */
  Serial.begin(9600);
}
void loop(){
  
}

void dumpMetrics(){  
  DynamicJsonDocument root(capacity);

  root["answer1"] = 12;
  root["answer2"] = 22;;
  root["answer3"] = 32;;

  
  serializeJson(root, Wire);
  serializeJson(root, Serial);
  Serial.println();
}

void requestEvent() {  
  dumpMetrics();
 }
```

the output from the master is: {"answer1":12,"answer2":22,"answ 
Which is good because I can confirm that they are able to communicate, while limited to the 32 bytes without increasing the buffer size. 
The bad thing being I clearly don't understand how to implement the library :sweat_smile:

mishave, 2020-07-21:
alrighty for those following..

On the master, i didn't put the `char json[capacity];` line right above the <DynamicJsonDocument doc(capacity);> line.

Still having the size issue, but as a workaround, I am splitting the message into three parts. my question is, is there a better way of doing this with the library?

Master:
```
#include <Wire.h>
#include <ArduinoJson.h>

const size_t capacity = JSON_OBJECT_SIZE(3)+20;

//I2C Slave Address
const int SensorBoardAdd = 8;
// I2C Commands
enum {
    CMD_SB01 = 1,
    CMD_SB02 = 2,
    CMD_SB03 = 3
    };

void setup(){
  Wire.begin(SensorBoardAdd); 
  Serial.begin(9600);
  Serial.println("setup::done");
}

void loop() {
  RequestSensorBoard();
  delay(50);
}

void RequestSensorBoard() {
  if (Wire.requestFrom (SensorBoardAdd, capacity) == 0)
    {
      //Use to throw an error need to think about this ;/
      }
    else  {
      sendCommand (CMD_SB01, capacity);
      char json[capacity];
      DynamicJsonDocument doc1(capacity);
      deserializeJson(doc1, Wire);
      int sb1 = doc1["s1"]; // 10
      int sb2 = doc1["s2"]; // 11
      int sb3 = doc1["s3"]; // 23
      Serial.println(" ");
      serializeJson(doc1, Serial);
    
      sendCommand (CMD_SB02, capacity);
      DynamicJsonDocument doc2(capacity);
      deserializeJson(doc2, Wire);
      int sb4 = doc2["s4"]; // 10
      int sb5 = doc2["s5"]; // 11
      int sb6 = doc2["s6"]; // 23
      Serial.println(" ");
      serializeJson(doc2, Serial);
   
      sendCommand (CMD_SB03, capacity);
      DynamicJsonDocument doc3(capacity);
      deserializeJson(doc3, Wire);
      int sb7 = doc3["s7"]; // 10
      int sb8 = doc3["s8"]; // 11
      int sb9 = doc3["s9"]; // 23
      Serial.println(" ");
      serializeJson(doc3, Serial);
    }
 
}

void sendCommand (const byte cmd, const int responseSize)
  {
  Wire.beginTransmission (SensorBoardAdd);
  Wire.write (cmd);
  Wire.endTransmission ();
  
  Wire.requestFrom (SensorBoardAdd, responseSize);  
  }
```

Slave:
```
#include <Wire.h>
#include <ArduinoJson.h>

char command=0;
const int SensorBoardAdd = 8;
enum {
    CMD_SB01 = 1,
    CMD_SB02 = 2,
    CMD_SB03 = 3
    };

void setup(){
  Wire.begin(SensorBoardAdd);
  Wire.onReceive(receiveEvent);
  Wire.onRequest(requestEvent);
  Serial.begin(9600);
}
void loop(){
  
}

void receiveEvent (int howMany)
  {
  command = Wire.read ();  // remember command for when we get request
  }

void requestEvent ()
  {
  switch (command)
     {
     case CMD_SB01:      dumpSB01(); break;  // send 1st Packet 
     case CMD_SB02:      dumpSB02(); break;  // send 2nd Packet
     case CMD_SB03:      dumpSB03(); break;  // send 3rd Packet
     }
  
  }

  void dumpSB01(){  
  const size_t capacity = JSON_OBJECT_SIZE(3);
  DynamicJsonDocument doc1(capacity);
  doc1["s1"] = 1200;
  doc1["s2"] = 2200;
  doc1["s3"] = 2300;
  serializeJson(doc1, Wire);
  serializeJson(doc1, Serial);
  Serial.println();
}

void dumpSB02(){  
  const size_t capacity = JSON_OBJECT_SIZE(3);
  DynamicJsonDocument doc2(capacity);
  doc2["s4"] = 5200;
  doc2["s5"] = 5200;
  doc2["s6"] = 5300;
  serializeJson(doc2, Wire);
  serializeJson(doc2, Serial);
  Serial.println();
}

void dumpSB03(){  
  const size_t capacity = JSON_OBJECT_SIZE(3);
  DynamicJsonDocument doc3(capacity);
  doc3["s7"] = 4240;
  doc3["s8"] = 4242;
  doc3["s9"] = 2214;
  serializeJson(doc3, Wire);
  serializeJson(doc3, Serial);
  Serial.println();
}
```

bblanchon, 2020-07-21:
Hi @mishave,

I'm pretty sure someone will be able to help you but, right now, there is too much information.
Can you summarize your question in a few lines?

Best regards,
Benoit

mishave, 2020-07-21:
@bblanchon thanks for that, I ended up answering my own original question in the end. The issue with both my code and issue #1099 forgetting to include in the master code: <char json[capacity];>

My question now is, is the ArduinoJSON Libary able to work around the i2c buffer size restriction or is just splitting it like above the generally accepted solution?

bblanchon, 2020-07-25:
Hi @mishave,

Please apologize for the delay; it took me some time to come up with an elegant solution.

I added a new decorator in the [StreamUtils](https://github.com/bblanchon/ArduinoStreamUtils) library for that.

> Wait and repeat write operations
> --------------------------------
> 
> Sometimes, a stream is limited to the capacity of its internal buffer. In that case, you must wait before sending more data.
> To solve this problem, StreamUtils provides the `WriteWaitingStream` decorator:
> 
> [image]
> 
> This function repeatedly waits and retries until it times out.
> You can customize the `wait()` function; by default, it's [`yield()`](https://www.arduino.cc/en/Reference/SchedulerYield).
> 
> For example, if you want to send more than 32 bytes with the [Wire library](https://www.arduino.cc/en/reference/wire), you can do:
> 
> ```c++
> WriteWaitingStream wireStream(Wire, [](){
>   Wire.endTransmission(false); // <- don't forget this argument
>   Wire.beginTransmission(address);
> });
> 
> Wire.beginTransmission(address); 
> wireStream.print("This is a very very long message that I'm sending!");
> Wire.endTransmission();
> ```
> 
> As you can see, we use the `wait()` function as a hook to flush the Wire transmission buffer. Notice that we pass `false` to [`endTransmission()`](https://www.arduino.cc/en/Reference/WireEndTransmission) so that it sends the data but doesn't actually stop the transmission.

For now, it's only available in the [WriteWaitingStream branch](https://github.com/bblanchon/ArduinoStreamUtils/tree/WriteWaitingStream); please check it out.
**I'm waiting for your feedback before merging it into the master branch.**

Best regards,
Benoit

mishave, 2020-07-27:
@bblanchon thank you very much for that, although not too sure how to implement your solution. as a test, I connected two Arduino's with a master writing the above code and the slave trying to capture it. couldn't understand how to use the StreamUtils to capture the message.

bblanchon, 2020-07-27:
The code above allows you to send a message larger than 32 bytes.
There is nothing special to do on the receiver side: simply call `deserializeJson(doc, Wire)` once to receive the complete document.

bblanchon, 2020-08-04:
I published [StreamUtils](https://github.com/bblanchon/ArduinoStreamUtils) 1.5.0 which includes `WriteWaitingStream`.

closed 2020-08-04 by bblanchon