bblanchon/ArduinoJson #1319: JsonVariant.set doesn't work in my fonction

sorokolokarim, 2020-07-23:
Hello ! Please help me to resolve this problem (sorry for my english)
I would "sample 2" work like "sample 1"

// sample 1 :

StaticJsonDocument<300> jsonDoc;

jsonDoc["hello"] = "world";

serializeJsonPretty(jsonDoc, Serial);

// Output :
//
// {
//    "hello" = "world"
// }

// sample 2 :

void variantSet(const JsonVariant variant, const char* value)
{
   variant.set(value);
}

StaticJsonDocument<300> jsonDoc;

variantSet(jsonDoc["hello"], "world");

serializeJsonPretty(jsonDoc, Serial);

// Output :
// NULL

bblanchon, 2020-07-23:
Hi @sorokolokarim,

[`JsonVarian::operator[]`](https://arduinojson.org/v6/api/jsonvariant/subscript/) doesn't return a `JsonVariant` but a `MemberProxy` or an `ElementProxy`.
These proxy classes allow inserting values in the document as you want.

When you cast an unbound `MemberProxy` (or `ElementProxy`) to `JsonVariant`, it returns null and doesn't modify the document.
It only changes the document if you perform a mutating operation, like `set()`, `to<T>()`, or `operator=()`.

That's why the following line works:

```c++
jsonDoc["hello"][0].set("world");
```

but the following doesn't:

```c++
variantSet(jsonDoc["hello"][0], "world");
```

To make this work, you can explicitly create the slot with `to<JsonVariant>()`, like so:

```c++
variantSet(jsonDoc["hello"][0].to<JsonVariant>(), "world");
```

See an online demo here: https://wandbox.org/permlink/zoSyyGkakUfluWxY

Best regards,
Benoit

sorokolokarim, 2020-07-25:
Thank you ! it's works.

closed 2020-07-25 by sorokolokarim