bblanchon/ArduinoJson #1320: How to create JSON in a loop?

HanSolo72, 2020-07-23:
In this [article](https://arduinojson.org/v5/faq/using-a-loop-why-are-values-all-the-same/) is explained how to create JSON in a loop, but this way doesn't work in V6.

The following code is running but still the same problem with always the same values:
```
#include` <ArduinoJson.h>

void setup() {
  Serial.begin(115200);
  const size_t capacity = JSON_ARRAY_SIZE(2) + JSON_OBJECT_SIZE(30) + 120;
  DynamicJsonDocument doc(capacity);
  JsonArray array = doc.to<JsonArray>();

  for (int i=0; i<3; i++) {
      char buffer[16];
      sprintf(buffer, "iteration %d", 0);
      array.add(buffer);
  }
  delay(2000);
  serializeJson(doc, Serial);

}

void loop() {
  // put your main code here, to run repeatedly:
}
```

Another question: How can I add pairs of values inside a loop? Espcially when both are dynamically? I want to parse a html form and save the values as JSON in a file.

bblanchon, 2020-07-24:
Hi @HanSolo72,

You made a typo while copying the snippet from the article.

Instead of:

```c++
sprintf(buffer, "iteration %d", 0);
```

you must write:

```c++
sprintf(buffer, "iteration %d", i);
```

Best regards,
Benoit

HanSolo72, 2020-07-24:
Hi Benoit,

thanks a lot, now it works!

closed 2020-07-24 by HanSolo72