bblanchon/ArduinoJson #1322: Merge creates null results when replacing keys

tavdog, 2020-07-25:
I'm trying to merge a remote json with a json pulled from spiffs on an ESP8266.  The spiffs json document is a static document of size 800,  the remote json is parsed to a dynamic json document of size 600.

The first json is the json living in SPIFFS.  The second json is the remote_json to merge into the first json.  Most of the data gets merged ok, but the api_data_url becomes null.  I can't figure out why this is happening.  When the devices reboots and tries a second time it works ok.   Sometimes  SPIFFS format will make it work the first time too.  Any ideas why this might be happening ?

json_config From SPIFFS : 
{
  "startup_message": "Kanaha",
  "display_type": "wf_wind",
  "brightness": "High",
  "api_token": "*******",
  "api_station_id": "166192",
  "api_data_url": "http://api.weatherflow.com/wxengine/rest/spot/getSpotDetailSetByList?units_wind=mph&spot_list=api_station_id&wf_token=",
  "update_interval_minutes": 5,
  "night_mode": "20,06,off",
  "update_seq": 2
}local ip

remote_json_config : 
{
    "update_seq":3,
    "startup_message":"Kanaha",
    "display_type":"wf_wind",
    "api_data_url":"http://api.weatherflow.com/wxengine/rest/spot/getSpotDetailSetByList?units_wind=mph&spot_list=api_station_id&wf_token=",
    "api_station_id":"166192",
    "update_interval_minutes":5,
    "night_mode":"20,06,dim"
}

Merge results : 
update_seq : 3
startup_message : Kanaha
display_type : wf_wind
**api_data_url : null**
api_station_id : 166192
update_interval_minutes : 5
night_mode : 20,06,dim

Here is the merge code : 
`//merge, 
    JsonObject object = remote_json_config.as<JsonObject>();
    for (auto kvp : object)
    {
        if (!kvp.value().isNull()) {
            json_config[kvp.key()] = kvp.value();
            //Serial.println(kvp.value().as<String>());
            Serial.print(kvp.key().c_str());Serial.println(" : "+json_config[kvp.key()].as<String>());
            if (json_config[kvp.key()].isNull()) {
                delay(1000);

                json_config["update_seq"] = 0;  // we have a null, so lets force the update for next time
                 //json_config[kvp.key()] = kvp.value();
            }

        }
    }`

bblanchon, 2020-07-25:
Hi,

The string is very long, so I suppose there wasn't enough space in the `JsonDocument` to copy it.
Try to increase the capacity of the `JsonDocument`.

It seems that both strings are identical, so you might try the `master` branch, which implements string deduplication, and therefore, would not need to make another copy of the string.

Lastly, remember that `JsonDocument` is designed to be a throw-away object; it's not meant to be kept alive for long.

See:

* [I found a memory leak in the library!](https://arduinojson.org/v6/faq/i-found-a-memory-leak-in-the-library/)
* [How to reuse a JsonDocument?](https://arduinojson.org/v6/how-to/reuse-a-json-document/)
* [Why must I create a separate config object? Why can't I use JsonDocument directly?](https://arduinojson.org/v6/faq/why-must-i-create-a-separate-config-object/)

Best regards,
Benoit

tavdog, 2020-07-25:
So when replacing the string there is a duplication occurring and so requires larger size to operate ?  That makes sense.  I've implemented a comparison so it will only try to replace if the data has changed.  Thanks again for a great library.

closed 2020-07-25 by tavdog