bblanchon/ArduinoJson #1323: How to MessagePack deserialize a very large document?

hm-3, 2020-07-26:
Here's how to work with large Json files. (https://arduinojson.org/v6/how-to/deserialize-a-very-large-document/)
Is there a way to do the same with large binary files and MessagePack format?
Thank you so much. This is a wonderful library.

bblanchon, 2020-07-26:
Hi @hm-3,

Currently, the filtering feature is only available for [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/).
If you need it for [deserializeMsgPack()](https://arduinojson.org/v6/api/msgpack/deserializemsgpack/), please put a thumb up on #1298.

You can, however, use the "deserialization in chunk" technique described in the article.
It works with both [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/) and [deserializeMsgPack()](https://arduinojson.org/v6/api/msgpack/deserializemsgpack/).

Best regards,
Benoit

hm-3, 2020-07-26:
Thank you for every second which spent on this library.
Your explanation was clear.

closed 2020-07-26 by hm-3