bblanchon/ArduinoJson #1324: How to access JSON object inside a loop?

HanSolo72, 2020-07-26:
Hi, I dynamically create a http form with 30 input elements and send them to the webserver by Fetch API. Then I save the data from the input elements as JSON to LittleFS. It all works fine, but after I read the saved JSON from LittleFS and want to build the website I want to read the saved values from the JSON object, but I don't know how to do this in a loop.

Here I build the 30 input elements (shortened, actually it's in a table):
```
  const int AnzahlReed=30;
  for(int i=1;i<=AnzahlReed;i++){
    sprintf(inputfield, "Reed-Kontakt %d: <input type=\"text\" name=\"reed%02d\" value=\"Test%d\">", i, i);
    server.sendContent(inputfield);
  }
```

The JSON looks like this: {"reed01": "Kitchen", "reed02": "Living room","reed03": "Main door"}

I just want to replace the text "Test" in the form with the value from the JSON.

I tried this inside the loop, but that doesn't work:
```
sprintf(reedNo, "reed%02d", i);
room = doc[reedNo];
Serial.print(room);
```

How can I read a value from the JSON an just put it into a variable?

bblanchon, 2020-07-27:
Hi,

The error says:

> You cannot assign `doc[reedNo]` to a variable of type `char[31]`"

Replace the following:

```c++
char room[31];
room = doc[reedNo];
```

with:

```c++
const char* room = doc[reedNo];
```

Best regards,
Benoit

HanSolo72, 2020-07-27:
Thank you, now it works perfectely.

closed 2020-07-27 by HanSolo72