bblanchon/ArduinoJson #1325: Calculating Size of JSON Payload (BODY)

Avong, 2020-07-26:
Hello please. 
What is the different between measurelength() and JSON_ARRAY_SIZE( ) + JSON_OBJECT_SIZE( )

I tried to investigate
To understand arduinoJson 6 library, out of curiosity, i needed JSON document of this structure
{
"device_id": 1,       
"location":   1,     
"floatvalue1":     1.1,
"floatvalue2":     1.2,
"floatvalue3":     1.3,
"status":      1,
"time": [ 23, 24, 12 ]
}

The arduino assistant gave as capacity JSON_ARRAY_SIZE(3) + JSON_OBJECT_SIZE(7) = 147
for AVR | 80+67 = 147

then i created sketch, in the loop i did something like this. 

**const size_t DATAsize = JSON_ARRAY_SIZE(3) + JSON_OBJECT_SIZE(7);
DynamicJsonDocument DATA(DATAsize);
    
  DATA["device_id"]=  1;
  DATA["location"]=   1;
  DATA["floatvalue1"]=    1.20;
  DATA["floatvalue2"]=    2.09;
  DATA["floatvalue3"]=    2.11;
  DATA["status"]=     1;
  JsonArray time = DATA.createNestedArray("time");
  time.add(23);
  time.add(24);
  time.add(12);

  size_t payloadsize = measureJson(DATA);
  
  debug.print("const size_t length= ");
  debug.println(DATAsize);
  debug.print("measureJson length= ");
  debug.println(payloadsize);
  delay(5000);**

This is the result i get. 
 
const size_t length= 80
measureJson length= 113

My thought the values should the same but that was wrong.
I need someone to help explain to me why these variation. 
If they are different, in calculating payload size for SIM800, which value should i use?
Is there any JSON library in C?

Meanwhile Mr bblanchon you have done great work. 

Thank you in advance.

bblanchon, 2020-07-27:
Hi,

[`JSON_OBJECT_SIZE()`](https://arduinojson.org/v6/api/jsonobject/) and [`JSON_ARRAY_SIZE()`](https://arduinojson.org/v6/api/jsonarray/) are macros that compute the size of the data structures in memory.
See also [JsonDocument::memoryUsage()](https://arduinojson.org/v6/api/jsondocument/memoryusage/)).

[`measureJson()`](https://arduinojson.org/v6/api/json/measurejson/) ([`measureLength()`](https://arduinojson.org/v5/api/jsonvariant/measurelength/), in version 5) returns the number of characters in the JSON output.
See also: [`measureJsonPretty()`](https://arduinojson.org/v6/api/json/measurejsonpretty/) and [`measureMsgPack()`](https://arduinojson.org/v6/api/msgpack/measuremsgpack/).

Best regards,
Benoit

Avong, 2020-07-28:
OK, 
Thanks that great. 
It means measureJson() and measurelength() can be use alternative to count number of characters in JSON for use in AT+HTTPDATA for SIM800.

closed 2020-08-04 by bblanchon