bblanchon/ArduinoJson #1326: ArduinoJson compilation error in ESP_IDF with arduino as components

manuprendlair, 2020-07-27:
Hello,

I am not able to compile your lib in an esp32 ESP-IDF project.
I got : 

 
[image]

At this step I could not find where is the problem.

Could You help ?

Emmanuel FIANCETTE
France

manuprendlair, 2020-07-27:
OK, I found in issue #1263 that I need to put flag ARDUINOJSON_ENABLE_ARDUINO_STRING to 1.
Now I got only 11 errors..

[image]

Could you help for these error ?

Thanks

manuprendlair, 2020-07-27:
my complete log is here : 

`C:\PROGRA~2\Sysprogs\VISUAL~1/ninja.exe 
[1/182] Performing build step for 'bootloader'
ninja: no work to do.
[2/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/cbuf.cpp.obj
[3/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-bt.c.obj
[4/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-gpio.c.obj
[5/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-adc.c.obj
[6/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-dac.c.obj
[7/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-cpu.c.obj
[8/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-sigmadelta.c.obj
[9/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-i2c.c.obj
[10/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-time.c.obj
[11/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-spi.c.obj
[12/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-psram.c.obj
[13/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-ledc.c.obj
[14/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-matrix.c.obj
[15/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/base64.cpp.obj
[16/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-misc.c.obj
[17/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-timer.c.obj
[18/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-touch.c.obj
[19/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/libb64/cdecode.c.obj
[20/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/libb64/cencode.c.obj
[21/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/HardwareSerial.cpp.obj
[22/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-rmt.c.obj
../../../components/arduino/cores/esp32/esp32-hal-rmt.c: In function 'rmtLoop':
../../../components/arduino/cores/esp32/esp32-hal-rmt.c:243:9: warning: unused variable 'channel' [-Wunused-variable]
     int channel = rmt->channel;
         ^
[23/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-uart.c.obj
[24/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/WMath.cpp.obj
[25/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/FunctionalInterrupt.cpp.obj
[26/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/Esp.cpp.obj
[27/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/main.cpp.obj
[28/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/MD5Builder.cpp.obj
[29/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/wiring_shift.c.obj
[30/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/IPv6Address.cpp.obj
[31/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/stdlib_noniso.c.obj
[32/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/IPAddress.cpp.obj
[33/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/wiring_pulse.c.obj
[34/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/libraries/BluetoothSerial/src/BluetoothSerial.cpp.obj
[35/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/StreamString.cpp.obj
[36/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/Print.cpp.obj
[37/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/Stream.cpp.obj
[38/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/libraries/DNSServer/src/DNSServer.cpp.obj
[39/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/libraries/EEPROM/src/EEPROM.cpp.obj
[40/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/WString.cpp.obj
[41/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/libraries/ESPmDNS/src/ESPmDNS.cpp.obj
[42/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/libraries/Preferences/src/Preferences.cpp.obj
[43/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/libraries/AsyncUDP/src/AsyncUDP.cpp.obj
[44/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/libraries/SimpleBLE/src/SimpleBLE.cpp.obj
[45/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/libraries/FS/src/FS.cpp.obj
[... cut ...] ic_istream<char>&}'
ninja: build stopped: subcommand failed.
-------------------------------------------------------------
Command exited with code 1
Executable: C:\PROGRA~2\Sysprogs\VISUAL~1/ninja.exe
Arguments: 
Directory: C:\Users\ManuF\source\repos\APEZD_Arduino_ESPIDF\test_Arduino_ESPIDF\test_Arduino_ESPIDF/build/VisualGDB/Debug
Command-line action failed

========== Project Build Summary ==========
    test_Arduino_ESPIDF  built in 02:21
========== Build: 0 Succeeded, 1 Failed ==========
`

manuprendlair, 2020-07-27:
complete log here : 

C:\PROGRA~2\Sysprogs\VISUAL~1/ninja.exe 
[1/182] Performing build step for 'bootloader'
ninja: no work to do.
[2/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/cbuf.cpp.obj
[3/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-bt.c.obj
[4/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-gpio.c.obj
[5/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-adc.c.obj
[6/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-dac.c.obj
[7/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-cpu.c.obj
[8/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-sigmadelta.c.obj
[9/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-i2c.c.obj
[10/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-time.c.obj
[11/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-spi.c.obj
[12/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-psram.c.obj
[13/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-ledc.c.obj
[14/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-matrix.c.obj
[15/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/base64.cpp.obj
[16/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-misc.c.obj
[17/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-timer.c.obj
[18/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-touch.c.obj
[19/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/libb64/cdecode.c.obj
[20/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/libb64/cencode.c.obj
[21/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/HardwareSerial.cpp.obj
[22/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-rmt.c.obj
../../../components/arduino/cores/esp32/esp32-hal-rmt.c: In function 'rmtLoop':
../../../components/arduino/cores/esp32/esp32-hal-rmt.c:243:9: warning: unused variable 'channel' [-Wunused-variable]
     int channel = rmt->channel;
         ^
[23/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/esp32-hal-uart.c.obj
[24/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/WMath.cpp.obj
[25/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/FunctionalInterrupt.cpp.obj
[26/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/Esp.cpp.obj
[27/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/main.cpp.obj
[28/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/MD5Builder.cpp.obj
[29/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/wiring_shift.c.obj
[30/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/IPv6Address.cpp.obj
[31/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/stdlib_noniso.c.obj
[32/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/IPAddress.cpp.obj
[33/180] Building C object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/wiring_pulse.c.obj
[34/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/libraries/BluetoothSerial/src/BluetoothSerial.cpp.obj
[35/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/StreamString.cpp.obj
[36/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/Print.cpp.obj
[37/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/Stream.cpp.obj
[38/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/libraries/DNSServer/src/DNSServer.cpp.obj
[39/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/libraries/EEPROM/src/EEPROM.cpp.obj
[40/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/cores/esp32/WString.cpp.obj
[41/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/libraries/ESPmDNS/src/ESPmDNS.cpp.obj
[42/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/libraries/Preferences/src/Preferences.cpp.obj
[43/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/libraries/AsyncUDP/src/AsyncUDP.cpp.obj
[44/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/libraries/SimpleBLE/src/SimpleBLE.cpp.obj
[45/180] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/libraries/FS/src/FS.cpp.obj
[... cut ...] asic_istream<char>&}'
ninja: build stopped: subcommand failed.
-------------------------------------------------------------
Command exited with code 1
Executable: C:\PROGRA~2\Sysprogs\VISUAL~1/ninja.exe
Arguments: 
Directory: C:\Users\ManuF\source\repos\APEZD_Arduino_ESPIDF\test_Arduino_ESPIDF\test_Arduino_ESPIDF/build/VisualGDB/Debug
Command-line action failed

========== Project Build Summary ==========
    test_Arduino_ESPIDF  built in 02:21
========== Build: 0 Succeeded, 1 Failed ==========

manuprendlair, 2020-07-27:
My complete compil log is here.
[log.txt](https://github.com/bblanchon/ArduinoJson/files/4982322/log.txt)

bblanchon, 2020-07-27:
Hi @manuprendlair,

> error: no matching function for call to 'makeReader(WiFiClient&)'

It looks like [ARDUINOJSON_ENABLE_ARDUINO_STREAM](https://arduinojson.org/v6/api/config/enable_arduino_stream/) is set to `0`.
Please try to set it to `1`.

Best regards,
Benoit

manuprendlair, 2020-07-27:
Yes Benoit ! you're the best!

ARDUINOJSON_ENABLE_ARDUINO_STREAM was on dependence of in an dependence of an #ifdef ARDUINO in Configuration.hpp....and of course ARDUINO is not define in my ESP-IDF project !

Merci !

closed 2020-07-27 by manuprendlair

bblanchon, 2020-07-27:
You're welcome @manuprendlair.
Good luck with your project.

manuprendlair, 2020-07-27:
;-)