bblanchon/ArduinoJson #1327: Visual Studio Intellisense String ambiguities caused by ARDUINOJSON namespace

MartinNohr, 2020-07-27:
I'm using Visual Micro in Visual Studio to develop ESP32 software.
I get undefined errors for json library objects like StaticJsonDocument. The code does compile, but code completion during editing fails due to these errors.
I can make it work by adding this line:
using namespace ARDUINOJSON_NAMESPACE;
But... now I get ambiguous definitions for any String objects. I suspect something is defining String in the json code, but I haven't figured it out yet. Any ideas would be appreciated.
Other than that the json library is wonderful! Thanks.

bblanchon, 2020-07-28:
Hi @MartinNohr,

This looks similar to #1181.
Please confirm.

Best regards,
Benoit

MartinNohr, 2020-08-01:
It is certainly similar. I'm getting "ambiguous definition" for String vars when I include the json namespace. I'll experiment with some of the suggestions and let you know what I figure out.

bblanchon, 2020-08-04:
@MartinNohr, could you confirm that this issue is a duplicate of #1181?

closed 2020-08-10 by bblanchon