bblanchon/ArduinoJson #1329: Crash when reading value that doesn't exists

HanSolo72, 2020-07-28:
Hi, I have another problem:

The problem is that unchecked checkboxes are not send by POST, so I have no value in the JSON object.

```
const char onValue[3] = "on";
  for(int i=1;i<=AnzahlReed;i++){
    sprintf(cbNo, "cbr%02d", i);
    const char* cbState = doc[cbNo];
    if (strcmp(cbState,onValue)==0){
      strcpy(cbState2,"checked");
    } else {
      strcpy(cbState2,"  ");
    }
```

This crashes when the pointer to cbState is null or empty. How can I check if there is no value in the JSON?

bblanchon, 2020-07-29:
Hi @HanSolo72,

You can safely write this:

```c++
if (doc[cbNo] == "on")
```

See also: [Why does my device crash or reboot?](https://arduinojson.org/v6/faq/why-does-my-device-crash-or-reboot/)

Best regards,
Benoit

HanSolo72, 2020-07-29:
Thank you, it's working now. I searched and googled for it but didn't find a solution. Much easier than I thought.

closed 2020-07-29 by HanSolo72

bblanchon, 2020-07-29:
I can add a page on arduinojson.org.
What words did you search for?