bblanchon/ArduinoJson #1330: Sometimes Strange Artifacts and incomplete Text

v11, 2020-07-28:
Hello Benoît

I am working now on a simple Message-Display.

1. ESP8266 connects to Wifi and my Webserver
2. Get JSON Data (https://www.vision11.ch/share/koni/data.json)
3. Display Message on E-Ink Display using GxEPD2 Library

To get the data, this works 100%. Right after the deserialization i log the message and it is 100% correct.
After that i start the rendering on the display. Sometimes it works and the string is complete. Sometimes i get wierd characters.

See Screenshot: http://vision11.ch/share/temp/strange-artifacts.PNG

Do you have an idea why this is? I also tried to implement the "WriteBufferingStream" but i am not sure if this is the way to go.

Do you have a tipp?

Have a nice day, 
Lukas

Code:
```// Wifi Setup: PreProcessor: Chech if SSID was already defined
#ifndef STASSID
#define STASSID "Lukas WiFi"
#define STAPSK "************"
// #define STASSID "INTERNETZ43"
// #define STAPSK "iuiuiuiu"
#endif

// Wifi: Library: esp8266 Wifi support. Based on WiFi.h from Arduino WiFi shield library.
#include <ESP8266WiFi.h>

// Wifi: Library: Variant of WiFiClient with Transport Layer Security (TLS) support. Deprecated predecessor: Secure Sockets Layer (SSL)
#include <WiFiClientSecure.h>

// NTP: Library: Library to get the current time. Needs internet connection.
#include <NTPClient.h>
#include <WiFiUdp.h>

// JSON: Library: Include JSON Library https://arduinojson.org
#include <ArduinoJson.h>
#include <StreamUtils.h>

// Wifi: Set WiFi Credentials, Host and Port (443)
const char *ssid = STASSID;
const char *password = STAPSK;
// const char *host = "www.strava.com";
// const int httpsPort = 443;

// Wifi: SHA1 fingerprint of the certificate, rendered as a hexadecimal number, 40 digits long
// Wifi: Use the web browser to show the certficate and read it from there
const char fingerprint[] PROGMEM = "1C 84 A1 6C E9 33 B1 2C 94 23 4D D5 69 EC 95 6A 53 68 82 69";

// GxEPD2: Base class GxEPD2_GFX can be used to pass references or pointers to the display instance as parameter, uses ~1.2k more code.
// GxEPD2: Enable or disable GxEPD2_GFX base class:
#define ENABLE_GxEPD2_GFX 0

// GxEPD2: Library: Arduino Display Library for SPI E-Paper Displays for black and white displays https://github.com/ZinggJM/GxEPD2
#include <GxEPD2_BW.h>
#include <GxEPD2_3C.h>

// Adafruit_GFX: Library: Adafruit_GFX library
#include <Adafruit_GFX.h>

// Adafruit_GFX: Fonts from Adafruit_GFX
#include <Fonts/FreeSans9pt7b.h>
#include <Fonts/FreeSans12pt7b.h>
#include <Fonts/FreeSans18pt7b.h>
#include <Fonts/FreeSans24pt7b.h>
#include <Fonts/FreeSansBold9pt7b.h>
[... 244 lines cut ...]
        display.println(message);

        // Last Update
        display.setFont(&FreeSans9pt7b);
        display.setCursor(610, marginTop + font9ptHeight);
        display.print("Updated: ");
        display.println(timeClient.getFormattedTime());

        // Draw Bike Icon
        // display.drawInvertedBitmap(40, 25, icnbike, 100, 100, GxEPD_BLACK);

    } while (display.nextPage());
}

void loop()
{
    // // Update every Hour
    // delay(3600000);

    // // Deep Sleep Mode (in Micro Seconds) We need to tie the RST pin to GPIO 16 on the ESP8266. On the NodeMCU, GPIO 16 is represented as D0.
    // //Serial.println("Starting Deep Sleep 30 Seconds");
    // ESP.deepSleep(30e6);
}

```

bblanchon, 2020-07-29:
Hi @v11,

It seems that the global variable `message` points to a destructed string in `doc`.
A simple solution is to change `message` to be a `String`, so that it implicitly makes a copy.

Also, it doesn't make sense to use `WriteBufferingStream` with `deserializeJson()`, because this decorator only affects the writing side of the stream. Instead, you should use `ReadBufferingStream`.

Best regards,
Benoit

v11, 2020-07-29:
Thank you. I could solve this now!

closed 2020-08-04 by bblanchon