bblanchon/ArduinoJson #1332: How can I remove empty object from an array in JSON

soloex1991eric, 2020-07-30:
I used the new json filter to filter an array of json objects and noticed that when the filter can't find the specified key it would leave in its place, an empty {}.
E.g: filtering,  `"LS": [{"Rd": {"Msg": "Hello"}},{"Bl":{"Msg":"Hello"}},{"Yl":{"Msg": "Hello"}}]` . 
Using, `filter["LS"][0]["Rd"]= true`.
Returns, `{"LS": [{"Rd": {"Msg": "Hello" }}, { "Rd": { "Msg": "Hello" } }, {}]} `. 
Is it possible to remove the empty {} ? 
I might be using the function completely wrongly, so any advice is welcomed.

bblanchon, 2020-07-30:
Hi @soloex1991eric,

This is the expected behavior: the filter doesn't change the number of elements in the array.

If you want to remove the empty elements, you must do it in your code.
For example, you could do:

```c++
JsonArray arr = doc["LS"];
for (JsonArray::iterator it = arr.begin(); it != arr.end(); ++it) {
  if (it->size() == 0)
    arr.remove(it);
}
```

Best regards,
Benoit

bblanchon, 2020-07-30:
PS: [demo on wandbox.org](https://wandbox.org/permlink/XvFtA2hqu85IAOG2)

soloex1991eric, 2020-07-30:
Thank you, that was what i was looking for.

closed 2020-07-30 by soloex1991eric

bblanchon, 2020-07-30:
You're welcome.
Good luck with your project, @soloex1991eric.