bblanchon/ArduinoJson #1334: Assistant is not giving correct sizing

davidkjackson54, 2020-07-31:
I used the ArduinoJson version 6 Assistant for this simple JSON: 
The assistant calculated Capacity as follows:

**const size_t capacity = JSON_OBJECT_SIZE(4);**
DynamicJsonDocument doc(capacity);

doc["device"] = "0";
doc["serial"] = "28dce056b5013cfc";
doc["c"] = 26.06;
doc["f"] = 78.91;

serializeJson(doc, Serial);

The size allocated to capacity is  128.
However with that size, only "service" and "serial" are output to Serial.
I had to change capacity to a higher value to get it to output all 4 fields. 
I had to increase the size of capacity to 148 before all 4 fields were output.

bblanchon, 2020-08-01:
Hi @davidkjackson54,

I copied the code above to wandbox, and it works as expected.
See by yourself: https://wandbox.org/permlink/PfHsJqtOHlux9B9L

I guess that your program uses `String` variables instead of string literals.
In that case, you must make room for them in the `JsonDocument`.
The assistant shows the required size under "Additional bytes for strings duplication".

Please see [Serialization tutorial](https://arduinojson.org/v6/doc/serialization/), section 4.6: "Duplication of strings".

Best regards,
Benoit

davidkjackson54, 2020-08-03:
Thanks for the clarification. Much appreciated.

closed 2020-08-03 by davidkjackson54

bblanchon, 2020-08-04:
You're welcome, @davidkjackson54.
Good luck with your project.