bblanchon/ArduinoJson #1335: DeserializeJson Issue when a key has an empty value '{}'

Pacheu, 2020-08-03:
Hi ! 
I am currently working on a ESP32 using the ESP-IDF SDK (latest version), and I have imported the ArduinoJSON library as a submodule in my git project and as a component of my IDF-project. I had a problem after using JsonDocument::deserializeJson(), which was that I only had access to the start of my parsed data, and after a certain point, everything was just not found and null. Moreover, the function was not returning any error, so I had no idea what was the problem.
I did not have any issue until last release (v6.16.0) 2 or 3 days ago. So I made some test with a function like this one :

void trySomething() {
	const char* TAG = "trySomething";
	String b("{\"output\":{},\"test\":\"25\"}"); //does not work, cannot acces to test
	String c("{\"output\":{\"output\":\"true\"},\"test\":\"25\"}"); //does work, CAN acces to test
	StaticJsonDocument<128> doc;
	DeserializationError error = deserializeJson(doc, b.c_str());
	if (error) {
		ESP_LOGE(TAG, "deserializeJson() 1 failed: %s", error.c_str());
		doc.clear();
		return ;
	}
	JsonObject obj = doc.as<JsonObject>();
	print_recursive(obj);
	if (doc["test"].isNull()) {
		ESP_LOGE(TAG, "test in b is null");
	}
	doc.clear();
	error = deserializeJson(doc, c.c_str());
	if (error) {
		ESP_LOGE(TAG, "deserializeJson() 2 failed: %s", error.c_str());
		doc.clear();
		return ;
	}
	obj = doc.as<JsonObject>();
	print_recursive(obj);
	if (doc["test"].isNull()) {
		ESP_LOGE(TAG, "test in c is null");
	}
}  witch uses this function : 
void print_recursive(JsonObject &obj) {
	const char* TAG = "print_recursive";
	for (JsonObject::iterator it=obj.begin(); it!=obj.end(); ++it) {
			ESP_LOGI(TAG, "%s", it->key().c_str());
			JsonObject obj_inside = obj[it->key().c_str()];
			if (obj_inside.begin()!= obj_inside.end()) {
				print_recursive(obj_inside);
			}
	}
}
It turns out that anything after an empty value '{}' of a key like "output" : {} will not be parsed, using v6.16.0. Screenshot :
[image]

Then, after switching back to v6.15.2, I can acces to the data after this "output" : {} of my var b ! Screenshot : 
[image]

I don't know exactly was could be the issue, but I just wanted to report what I experienced.

bblanchon, 2020-08-04:
Hi @Pacheu,

Thank you very much for reporting this issue.
I don't understand how this could pass the test suite 😲 
I'll fix this ASAP.

Best regards,
Benoit

closed 2020-08-04 by bblanchon

bblanchon, 2020-08-04:
The fix is available in [ArduinoJson 6.16.1](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.16.1).

commit 35a39b8d (2020-08-04): Fixed `deserializeJson()` that stopped reading after `{}` (fixes #1335)