bblanchon/ArduinoJson #1337: String and char produce strange output on ESP32 or ESP8266

hdse, 2020-08-06:
Hi,
I have a strange problem with your standard Deserialization tutorial.
I use Arduino IDE 1.8.13
ArduinoJson 6.16.1

**It works fine on an Arduino Uno**, output correct

gps
1351824120
48.756081
2.302038

**but not on ESP** ouput wrong

ESP produce allways  strange sings befor the correct value. Tested on different ESP boards

!,▒▒bf4HXa>@%▒gps
1351824120
48.756081
2.302038

Any solutions or hints for me

Thanks Peter

uski, 2020-08-07:
Have you checked this ?
https://arduinojson.org/v6/faq/why-does-the-output-contain-garbage/

bblanchon, 2020-08-07:
Hi Peter,

There are always some random bytes when you boot an ESP; this is not related to ArduinoJson.

That's because the ESP prints some information at a different baud rate before executing `Serial.begin(9600)`.
From what I saw, the initial baud rate of the ESP is 74880 bps, but it might differ from one board to another.

When I set the baud rate to 74880 in the program and in the Serial Monitor, I see the following output:

```

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld
gps
1351824120
48.756081
2.302038
```

Best regards,
Benoit

bblanchon, 2020-08-07:
PS: I added [a dedicated section](https://arduinojson.org/v6/faq/why-does-the-output-contain-garbage/#esp-garbage) in the article mentioned by @uski.

closed 2020-08-14 by bblanchon