bblanchon/ArduinoJson #1338: Update JSON Document

regina2898, 2020-08-07:
Hi, so I have 2 transmitter and receiver Arduino HC-12 code 

This is the sender code
```
#include <SoftwareSerial.h>
#include <ArduinoJson.h>
#define rxPin 5
#define txPin 4
#define setPin 6

SoftwareSerial HC12(rxPin, txPin); // ke arah sensor node

void setup() {
  Serial.begin(19200);
  pinMode(rxPin, INPUT);
  pinMode(txPin, OUTPUT);
  HC12.begin(9600);
  if (HC12.isListening()) {
    Serial.println("OK");
  } else {
    Serial.println("NOK");
  }
}

void loop() {
  while (!Serial) continue;
  StaticJsonDocument<200> doc;

  doc["sessionid"] = 1;
  doc["uplink"] = false;

  JsonArray idmeter = doc.createNestedArray("idmeter");
  idmeter.add(1);
  idmeter.add(2);
  idmeter.add(3);
  JsonArray standmeter = doc.createNestedArray("standmeter");
  standmeter.add(0);
  standmeter.add(0);
  standmeter.add(0);

  serializeJsonPretty(doc, HC12);

  delay(300);
  void clear();
  delay(1000);

  StaticJsonDocument<200> dok;
  deserializeJson(dok, HC12);
  serializeJsonPretty(doc, Serial);
}
```
And this is the receiver code :
```
#include <SoftwareSerial.h>
#include <ArduinoJson.h>
#define rxPin 5
#define txPin 4
#define setPin 6

SoftwareSerial HC12(rxPin, txPin); // ke arah sensor node
void setup() {
  // Initialize serial port
  Serial.begin(9600);
  while (!Serial) continue;
  pinMode(rxPin, INPUT);
  pinMode(txPin, OUTPUT);
  HC12.begin(9600);
  if (HC12.isListening()) {
    Serial.println("OK");
  } else {
    Serial.println("NOK");
  }
}

void loop() {
  StaticJsonDocument<200> doc;
  DeserializationError error = deserializeJson(doc, HC12);
  if (error) {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.c_str());
    return;
  }
  int indeks;
  int sessionid = doc["sessionid"];
  boolean uplink = doc["uplink"];
  JsonArray idmeter = doc["idmeter"].as<JsonArray>();
  JsonArray standmeter = doc["standmeter"].as<JsonArray>();
  int count = idmeter.size();

  if (uplink == false) {
    uplink = true;
    for (int i = 0; i < count; i++) { //Iterate through results
      doc["idmeter"][i];  //Implicit cast
      if (doc["idmeter"][i] == 2) {
        indeks = i;
        doc["standmeter"][indeks] = 100;
      }
    }
  }

  delay(300);
  serializeJsonPretty(doc, HC12);
}
```

So the receiver send it back to my sender Arduino, then serial.print it
I wish to see updated JSON Data from the receiver, but it shows the old JSON Data from the sender

Does anyone know what might be wrong from my code?

Thanks before!

bblanchon, 2020-08-07:
Hi @regina2898,

Take a closer look at these two lines:

```c++
deserializeJson(dok, HC12);
serializeJsonPretty(doc, Serial);
```

Best regards,
Benoit

regina2898, 2020-08-07:
Hi, Benoit. I've tried typing dok, but it returns null.
Anything could be wrong?
Thanks before!

regina2898, 2020-08-07:
> Hi @regina2898,
> 
> Take a closer look at these two lines:
> 
> ```c++
> deserializeJson(dok, HC12);
> serializeJsonPretty(doc, Serial);
> ```
> 
> Best regards,
> Benoit

Hi, THanks benoit. I've realized it and have typed dok, but it returns NULL. what could be wrong?
Thanks before!

bblanchon, 2020-08-08:
Hi @regina2898,

Please check the [DeserializationError](https://arduinojson.org/v6/api/misc/deserializationerror/) returned by [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/).

Best regards,
Benoit

closed 2020-08-16 by bblanchon