bblanchon/ArduinoJson #1339: [ArduinoJson6] Issue while copyArray between JsonArray and const uint8_t*

gnalbandian, 2020-08-07:
Hi everyone.

I'm building a program with an ESP8266 and ArduinoJson library.
I'm having trouble to use the copyArray function. The compiler returns:

`ArduinoJson/Array/Utilities.hpp:35:8: error: request for member 'add' in 'dst', which is of non-class type 'const 
unsigned int'
     ok &= dst.add(src[i]);`

I have solved the issue by doing the copy of elements one by one. i.e:
`JsonArray MainAddr = root.createNestedArray("address");
for(uint8_t cnt2 = 0; cnt2 < sizeof(DevAddr); cnt2++)
                MainAddr.add(_customObj.getAddr()[cnt2]);`

bblanchon, 2020-08-07:
Hi @gnalbandian,

Check [the documentation](https://arduinojson.org/v6/api/misc/copyarray/) and you'll find the signature:

```c++
// 1D array -> JsonArray
bool copyArray(T* src, size_t len, JsonArray dst);
```

As you can see, the size is the second argument, not the third.

Best regards,
Benoit

gnalbandian, 2020-08-07:
Oh men, sorry about that.

I read the docs, but evidently, didn't read it as careful as needed.

Thanks @bblanchon

closed 2020-08-07 by gnalbandian