bblanchon/ArduinoJson #1340: Non valid json after deserialize()

sblantipodi, 2020-08-09:
Hi,
writing here to ask for help, hope that someone can help.

I'm using the latest arduinojson with pubsubclient
this is my mqtt callback

```
void callback(char* topic, byte* payload, unsigned int length) {

    StaticJsonDocument<BUFFER_SIZE> jsonDoc;

    if(strcmp(topic, STREAM_TOPIC) == 0) {

      deserializeJson(jsonDoc, payload);
      int lednum = jsonDoc["lednum"];
      JsonArray stream = jsonDoc["stream"];
     
      for (uint8_t i = 0; i < lednum; i++) {
      //do things
      }
 
    } else if (variable != "string") {

      deserializeJson(jsonDoc, payload);

      if (strcmp(topic, SMARTOSTAT_CLIMATE_STATE_TOPIC) == 0) {
        processSmartostatClimateJson(jsonDoc);
      } else if (strcmp(topic, CMND_AMBI_REBOOT) == 0) {
        processGlowWormLuciferinRebootCmnd(jsonDoc);
      } else if (strcmp(topic, LIGHT_SET_TOPIC) == 0) {
        processJson(jsonDoc);
      }
}
```

this callback call a function based on the topic. 

the processJson function for example does only this:
```
cons
  bool processJson(StaticJsonDocument<BUFFER_SIZE> json) {
      if (json.containsKey("effect")) {
        effect = json["effect"];
      }
  }
```
the variable effect is defined in my header file like this:
const char* effect = "solid";

it works, but sometimes when I print the variable effect I get a "chunked" JSON from a previous MQTT message from another topic. how this is possible? the variable effect is assigned only in the processJson function.
why I have dirty data in it?

Thanks

bblanchon, 2020-08-09:
Hi @sblantipodi,

```c++
effect = json["effect"];
```

This line saves a pointer to a string located in the `payload` buffer.
(That's because you use the zero-copy mode; otherwise, it would be located in the `JsonDocument`).

To fix this issue, you can copy the string. For example

```c++
// change the declaration of effect 
char effect[16] = "solid";

// copy the string
strncpy(effect, json["effect"], 16);
```

However, I presume that you only accept a limited number of effects.
If that's the case, an enum would be more appropriate.

```c++
// change declaration of effect
enum class Effect { solid, dashed, dotted };
Effect effect;

// assign the value
JsonVariant requestedEffect = json["effect"];
if (requestedEffect == "solid") effect = Effect::solid;
else if (requestedEffect == "dashed") effect = Effect::dashed;
else if (requestedEffect == "dotted") effect = Effect::dotted;
// else -> effect not supported
```

Now, if you have a lot of effect, you can use a loop:

```c++
// add a new array that contains the mapping of strings<->Effect
struct EffectString { Effect effect; const char* str; };
EffectString effectStrings[] = {
  Effect::solid, "solid",
  Effect::dashed, "dashed",
  Effect::dotted, "dotted"
  // ...more effects...
};

// assign the value
JsonVariant requestedEffect = json["effect"];
for (auto es : effectStrings) {
  if (requestedEffect == es.str) {
    effect = es.effect;
    break;
  }
}
```

Well, you get the idea.

Now, there is one last thing I need to say about your code.
You should not pass the `JsonDocument` by value, because it makes a useless copy of the document.
Instead, pass by reference.
Even better, pass by *const* reference, since you don't need to modify the document:

```c++
bool processJson(const JsonDocument& json)
```

If you need a refresher on references and strings, please check out "The Missing C++ Course" in [Mastering ArduinoJson](https://arduinojson.org/book/).

Best Regards,
Benoit

sblantipodi, 2020-08-09:
@bblanchon you are an example on how to drive an open source project.
bought the book to thank you for your work here.
thanks.

closed 2020-08-09 by sblantipodi

bblanchon, 2020-08-10:
Thank you very much, Davide. 🙏